teslamotors/fleet-telemetry #41: How to "share a vehicle configuration with Tesla"?

Niek, 2023-06-29:
Great project!

The README is pretty self-explanatory, but step 6 is a bit vague:

https://github.com/teslamotors/fleet-telemetry/blob/2cb04c30d25fcf643ed34dc96f4627dcee588c7a/README.md?plain=1#L114-L127

How can this config be shared? Is there some site where interested service providers can sign up/upload the config?

aaronpkahn, 2023-06-29:
Great question @Niek. More info to come on this soon.

stx, 2023-06-30:
@aaronpkahn Creator of Tessie here - this looks incredible and is super exciting. Thank you for supporting this and all of the work you and your team do. ❤️

ProductCrazy, 2023-06-30:
The doors this will open up are truly endless. Hoping other manufacturers follow suit. Can you talk through the limitations (if any) when it comes to use-cases for this data access?

Or is the intention behind the open concept that if the user (owner) deems the 3rd party has value and decides to share there's no limitations?

beaylott, 2023-07-03:
@aaronpkahn can i put a vote in for an API endpoint using current auth used by app to set this up?

ErikDeBruijn, 2023-07-12:
This is a really great development!

I'm also very interested in this, specifically how to become authorised to receive someone's vehicle telemetry. We're offering a smart charging service to allow people to plan their charging to occur especially when curtailment is happening on the grid ( Stekker.app ). We want to scope down the information we have access to, to a very small subset of what is being offered here, (e.g. only charging related data).

(on a nitpicking side-note: the file name `SECURTIY.md` has a typo)

Niek, 2023-07-19:
> @Niek any luck with this?
No, we're waiting on @aaronpkahn to provide details. I guess Tesla will launch this soon - maybe in tanden with the Q2 earnings call today?

tushar-mohan, 2023-07-20:
I also wanted clarification on the "ca" field in the client config. It would seem that if Tesla CA is issuing certificates to the device and the server, the CA cert would not need to be provided in a production scenario.

tonychill, 2023-07-20:
Whoever provides the CA (you or Tesla) would also have to create the client and server certs. With the scenario of the vehicle sending data to your prod env I would prefer to manage those certs.

tushar-mohan, 2023-07-20:
@tonychill thanks for reply. Yes, the same CA would need to issue both the certs -- and better it be in-house. Thanks for clarifying that. I was mistaken regarding the workflow earlier.

KartikAiyer, 2023-07-30:
I'm curious if this will also provide some of the remote access to things like locks and air conditioning? Fleet managers would be able to provide their drivers with some of the remote access that native tesla owners have like turning on the AC before starting a trip.

beaylott, 2023-08-24:
@aaronpkahn any updates on this?

aaronpkahn, 2023-08-24:
Not yet @beaylott.  We're currently in an active development and trial phase, and will communicate more when this phase completes.

tlbdk, 2023-08-24:
@aaronpkahn any way to sign up for the trial, we have quite a few customers who are interested in this?

aaronpkahn, 2023-08-24:
@tlbdk not quite yet. We need to first complete some infrastructure development to reduce the internal cost of onboarding fleets.

Meowcat285, 2023-10-03:
Any updates on this?

Niek, 2023-10-04:
https://www.tesla.com/en_gb/fleet#contact

I assume this is the way to register. As far as I can see, this is mostly useful for larger companies.

jdarpinian, 2023-10-12:
Registration is live now along with documentation https://developer.tesla.com/docs/fleet-api

slashmili, 2023-10-19:
We have setup the new fleet-api, we are getting vehicle data via REST but still not sure how can we share this file with Tesla. 

There is a section about [registering partner account](https://developer.tesla.com/docs/fleet-api#register), is this how we should share this info with Tesla? There is no mention of telemetry on that api doc and I'm not sure how to move forward with streaming data

vbarrier, 2023-10-25:
+1 release the kraken to stream data to our telemetry servers :-)

wesley-coxauto, 2023-11-07:
We are also confused about this. Any updates?

jlestel, 2023-11-14:
Please add WIP or update your API. Atm, we just lost time here.

hullo-rikard, 2023-11-16:
@aaronpkahn 👋 We started a small driving journal just a few weeks before the announcement that version 2023.38+ would no longer return any location data. We quickly implemented this new fleet-telemetry to be able to get location data again and are now stuck with angry users and now here to put our client/vehicle config 😓 Any type of update or information regarding this issue?

slashmili, 2023-11-16:
> @aaronpkahn 👋 We started a small driving journal just a few weeks before the announcement that version 2023.38+ would no longer return any location data. We quickly implemented this new [...]

Are you using `fleet-telemetry` or the REST API?

If you are using REST, there has been an update https://developer.tesla.com/docs/fleet-api#vehicle_data

> Makes a live call to the vehicle. This may return cached data if the vehicle is offline. For vehicles running firmware versions 2023.38+, **location_data** is required to fetch vehicle location. This will result in a location sharing icon to show on the vehicle UI.

If you are using `fleet-telemetry` I'm eager to learn how did you share the config with Tesla and the car

mittsh, 2023-11-30:
@aaronpkahn Regarding _Fleet Telemetry_, do you have any update on the last step _"Create and share a vehicle configuration with Tesla"_? How to share (or upload, or POST) the file `client_config.json`?

Also curious if you got this to work @slashmili?

p.s. really appreciate the direction the Fleet API is taking, this is awesome, huge thanks to the team!

slashmili, 2023-11-30:
@mittsh nope! Still waiting for more information to put fleet-telemetry in use

slashmili, 2023-12-11:
FYI there has been an update in Tesla's Developer Doc regarding sharing the config:

https://developer.tesla.com/docs/fleet-api#fleet-telemetry

agbpatro, 2024-01-03:
We have updated instructions and documentation regarding this [here](https://developer.tesla.com/docs/fleet-api#2024-01-02-adding-support-for-registering-vehicles-for-fleet-telemetry)

Bre77, 2024-01-03:
> We have updated instructions and documentation regarding this [here](https://developer.tesla.com/docs/fleet-api#2024-01-02-adding-support-for-registering-vehicles-for-fleet-telemetry)

Thats great to hear @agbpatro, are Fleet API Support still responding to CSRs over the holiday period? The new API is working in so far as It does not like my self-signed certificate 😄

agbpatro, 2024-01-03:
> Thats great to hear @agbpatro, are Fleet API Support still responding to CSRs over the holiday period? The new API is working in so far as It does not like my self-signed certificate 😄

Yes we are incrementally ramping up the process for responding to CSR requests

closed 2024-01-11 by agbpatro