teslamotors/fleet-telemetry #76: How to configure the server keys?

slashmili, 2023-11-04:
I have problem receiving vehicle data in my server. I'm not sure if it's the server key config issue or it's something else.

Anyhow I'd like to go through the steps I took and the result I'm experiencing and hopefully you can point me to the right direction:

## Steps:
I have followed the guides based on the [fleet docs](https://developer.tesla.com/docs/fleet-api)

### 1. Register domain
I create the keys: 
```
openssl ecparam -name prime256v1 -genkey -noout -out private-key.pem
openssl ec -in private-key.pem -pubout -out com.tesla.3p.public-key.pem

openssl req -new -x509 -key private-key.pem -out client-certificate.pem
```
I have registered a partner account domain: 

I submitted `client-certificate.pem` as `ca` and host `com.tesla.3p.public-key.pem` as `https://mysubdomain.high-mobility.com/.well-known/appspecific/com.tesla.3p.public-key.pem`

```
curl https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts --data '{...}'

{
    "response": {
        "client_id": '....',
        "domain": "mysubdomain.high-mobility.com",
        "ca": "-----BEGIN CERTIFICATE-----\n....",
        "public_key": "04418....b3c9",
   }
}
```

### 2. Distribute key in the car.

I also followed the stepped [Distributing your public key](https://github.com/teslamotors/vehicle-command#distributing-your-public-key) in  Vehicle Command SDK repo and the vehicle as my "Fleet Key" 

[image]

### Question
So the question that I have is how to configure `server_config.json` ?

This is my current config:
```
{
  "host": "0.0.0.0",
  "port": 443,
....
  "records": {
   ...
   "V": ["logger", "kafka"]
  },
  "tls": {
      "server_cert": "/etc/certs/server/tls.crt",
      "server_key": "/etc/certs/server/tls.key"
  }
}
```
the `tls.crt` and `tls.key` are valid certificates I obtained for this domain from GlobalSign.

However I see that it doesn't work. No data in log nor in Kafka, the LB is receiving many request while I'm driving the car but all I see in the `fleet-telemtey` log is TLS errors.

Do I configure the server with wrong keys?

#### Notes
1. I can confirm the TLS terminates on my `fleet-telemetry` instance
2. I can verify the certificates are working because running 
```
openssl s_client -connect mysubdomain.high-mobility.com:443 -servername mysubdomain.high-mobility.com -showcerts 

CONNECTED(00000006)
depth=2 OU = GlobalSign Root CA - R3, O = GlobalSign, CN = GlobalSign
verify return:1
depth=1 C = BE, O = GlobalSign nv-sa, CN = GlobalSign RSA OV SSL CA 2018
verify return:1
depth=0 CN = *.high-mobility.com, O = High-Mobility GmbH, L = Berlin, ST = Berlin, C = DE
verify return:1
---
Certificate chain
 0 s:CN = *.high-mobility.com, O = High-Mobility GmbH, L = Berlin, ST = Berlin, C = DE
   i:C = BE, O = GlobalSign nv-sa, CN = GlobalSign RSA OV SSL CA 2018
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: Sep  5 09:12:47 2023 GMT; NotAfter: Oct  6 09:12:46 2024 GMT

-----BEGIN CERTIFICATE-----
....
---
Server certificate
subject=CN = *.high-mobility.com, O = High-Mobility GmbH, L = Berlin, ST = Berlin, C = DE
issuer=C = BE, O = GlobalSign nv-sa, CN = GlobalSign RSA OV SSL CA 2018
---
Acceptable client certificate CA names
CN = Tesla Issuing CA, O = Tesla Motors, L = Palo Alto, ST = California, C = US
CN = Tesla Motors GF Austin Product Issuing CA, OU = Motors, OU = PKI, O = Tesla Inc., C = US
CN = Tesla Motors GF Berlin Product Issuing CA, OU = Motors, OU = PKI, O = Tesla Inc., C = US
CN = Tesla Motors GF0 Product Issuing CA, OU = Motors, OU = PKI, O = Tesla Inc., C = US
CN = Tesla Motors GF3 Product Issuing CA, OU = Motors, OU = PKI, O = Tesla Inc., C = US
CN = Tesla Motors GF3 Product RSA Issuing CA, OU = Motors, OU = PKI, O = Tesla Inc., C = US
CN = Tesla Motors Product Issuing CA, OU = Motors, OU = PKI, O = Tesla Inc., C = US
CN = Tesla Motors Product RSA Issuing CA, OU = Motors, OU = PKI, O = Tesla Inc., C = US
CN = Tesla Motors Products CA
CN = Tesla Motors Root CA
CN = Tesla Policy CA, O = Tesla Motors, L = Palo Alto, ST = California, C = US
CN = Tesla Product RSA Root CA, OU = PKI, O = Tesla, C = US
CN = Tesla Product Root CA, OU = PKI, O = Tesla, C = US
CN = Tesla Root CA, O = Tesla Motors, L = Palo Alto, ST = California, C = US
Requested Signature Algorithms: RSA-PSS+SHA256:ECDSA+SHA256:Ed25519:RSA-PSS+SHA384:RSA-PSS+SHA512:RSA+SHA256:RSA+SHA384:RSA+SHA512:ECDSA+SHA384:ECDSA+SHA512:RSA+SHA1:ECDSA+SHA1
Shared Requested Signature Algorithms: RSA-PSS+SHA256:ECDSA+SHA256:Ed25519:RSA-PSS+SHA384:RSA-PSS+SHA512:RSA+SHA256:RSA+SHA384:RSA+SHA512:ECDSA+SHA384:ECDSA+SHA512
Peer signing digest: SHA256
Peer signature type: RSA-PSS
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 4798 bytes and written 426 bytes
Verification: OK
---
New, TLSv1.3, Cipher is TLS_AES_128_GCM_SHA256
Server public key is 2048 bit
This TLS version forbids renegotiation.
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---
402B4345F87F0000:error:0A000412:SSL routines:ssl3_read_bytes:sslv3 alert bad certificate:ssl/record/rec_layer_s3.c:1586:SSL alert number 42
```

davidlin2k, 2023-11-14:
Any luck with share a vehicle configuration with Tesla?

slashmili, 2023-11-14:
Short answer No.

I shared my `client_config.json` with Tesla Fleet support and I was instructed to take this approach instead which I did. However I've come to the conclusion that this is not the way to share the 3rdparty fleet-telemetry server with the car.

davidlin2k, 2023-11-14:
> Short answer No.
> 
> I shared my `client_config.json` with Tesla Fleet support and I was instructed to take this approach instead which I did. However I've come to the conclusion that this is not the way to share the 3rdparty fleet-telemetry server with the car.

Thanks for the update, I guess we will have to wait for more instructions

nitishsingla91, 2023-11-15:
@slashmili 

I am also experiencing TLS Handshake Error, Have you get success with this step?

2023/11/15 22:48:22 http: TLS handshake error from 100.94.115.106:38807: EOF 

Could you provide guidance on troubleshooting at least for this step?

slashmili, 2023-11-16:
@nitishsingla91 From my understand that is meant to be like that!

As mentioned in the [readme](https://github.com/teslamotors/fleet-telemetry#configuring-and-running-the-service), this server relies on mTLS. Only clients that have "Tesla" certificates can make communication to this server.

nitishsingla91, 2023-11-22:
@slashmili 
I have one more question for you. You said in your first comment that you submitted **client-certificate.pem** as the **CA**. 
Where did you submitted it exactly? 

I followed your steps, created the KEYS, and then created the partner token. But when I tried to call the Partner Account Endpoint with the same token, I got this error:
`{
    "response": null,
    "error": "https://auth.tesla.com:443/oauth2/v3/clientinfo => operation_timedout with 7s timeout",
    "error_description": ""
}`

slashmili, 2023-11-22:
@nitishsingla91 I did using this api 
```
curl https://fleet-api.prd.<region: na|eu>.vn.cloud.tesla.com/api/1/partner_accounts \
    -h 'Authorization: Bearer <PARTNER-JWT-TOKEN>` \
    --data '{"domain": "<your-domain>", "ca": "-----BEGIN CERTIFICATE-----\n...."}'

{
    "response": {
        "client_id": '....',
        "domain": "<your-domain>",
        "ca": "-----BEGIN CERTIFICATE-----\n....",
        "public_key": "04418....b3c9",
   }
}
```

It was working when I tried!

Make sure your domain is publicly available servers the public key under `https://<your-domain>/.well-known/appspecific/com.tesla.3p.public-key.pem`

nitishsingla91, 2023-11-22:
@slashmili 
It appears that the contract has been updated; 
https://developer.tesla.com/docs/fleet-api#register
 
They are now accepting only one parameter:

`curl --header 'Content-Type: application/json' \
  --header "Authorization: Bearer $TESLA_API_TOKEN" \
  --data '{"domain":"string"}' \
  'https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts' 
`

slashmili, 2023-12-04:
> does that mean your vehicle was sending telemetry to your load balancer at mysubdomain.high-mobility.com?

That was my feeling! because there were so many parts in the middle, I was not able to debug further(or rather didn't want :D) 

> How did you configure your vehicle to start sending telemetry?
> Did you:
> 1. send your client_config.json to Tesla
> 2. install your public key (virtual key) on the vehicle as per [this instruction](https://github.com/teslamotors/vehicle-command#distributing-your-public-key)

1. Actually I did sent my `client_config.json` to [Tesla](https://developer.tesla.com/docs/fleet-api#help-and-support)! However I got a reply back: 
> third party apps can and should use the mobile app pairing flow documented in the [Vehicle Command](https://github.com/teslamotors/vehicle-command) SDK
2. Yes I installed it that way.

Initially I thought these are the same but I think this is not the case. 

1. If I start `fleet-telemetry` project, there is no option(as of today) to put the public key in `/.well-known/appspecific/com.tesla.3p.public-key.pem`
2. Even if I patch the code to accept the public key, when I call [register](https://developer.tesla.com/docs/fleet-api#register) API, I get an error that the Tesla server was not able to fetch the public key from my server because of mTLS.