teslamotors/fleet-telemetry #98: Which port is used for telemetry?

jbanyer, 2024-01-21:
Which network port should the telemetry service (load balancer) be listening on?

The setup [instructions](https://github.com/teslamotors/fleet-telemetry?tab=readme-ov-file#install-manually-skip-this-if-you-have-installed-with-helm-on-kubernetes) do not specify a port, and the [config](https://developer.tesla.com/docs/fleet-api#fleet_telemetry_config) endpoint does not mention supplying a port.

Bre77, 2024-01-21:
The setup instructions do specify a port, its literally `port` in the server config.json

As for the vehicle config, I was under the assumption `config.hostname` could include a port. For example `test-telemetry.com:8443`. If this is not the case I am will be very interested because I was somewhat depending on this.

patrickdemers6, 2024-03-02:
Changes went out earlier this week that fix the issue of port not being used.

closed 2024-03-02 by patrickdemers6