teslamotors/fleet-telemetry #101: How to confirm vehicles are streaming telemetry

bgk-, 2024-01-25:
We've setup our fleet server, supplied the CSR to the fleet support email and received confirmation, and can now register vehicles (and confirmed the tokens include the fleet urls in the `aud` array). But no data or requests are ever coming to our server.

Wondering if there is anything else we need to do, or anyway we can check if data is even trying to be sent.

Thanks!

wcoekaer, 2024-01-25:
Curious - how do you know the server is running -and- listening on the right port? Did you do a netstat or telnet to the port to see if something is listening? Do you run it on port 443? or a separate port? Do you provide a url or host name and port when you call the register vehicle? Did you upload the key to the vehicle? Not sure if that is a mandatory step as well, the instructions are a bit scattered around for the various parts. hopefully once a few folks have it working end to end we can have a quick start guide or so worked on as a contribution that explains all components and a diagram or so. 

How long did it take for Tesla to respond to your csr email? I sent a mail last week. I saw that it's responded to in order but so far nothing yet.

Bre77, 2024-01-25:
> How long did it take for Tesla to respond to your csr email? I sent a mail last week. I saw that it's responded to in order but so far nothing yet. 
> 

I sent my CSR January 1st and still don't have a response, so I'm also curious about that.

bgk-, 2024-01-25:
- I submitted my CSR email on January 5th
  - I have sent a config command to a vehicle successfully with a response: `{ response: { updated_vehicles: 1 } }`
  - I'm unsure if the server is listening on the correct port, however it's behind an ingress-nginx and I've set the suspected ports, and the logs show no requests to the ingress to setup a connection.
   - I'm only calling the register config with a hostname since that's what the docs seem to have, but I'll try with a few ports as well.
    - I haven't uploaded the key to the vehicle, from my understanding that isn't necessary.
    
  Hope that info helps. I'll still be testing out a few things and if I get anything working I'll be sure to update.
  
  Cheers!

agbpatro, 2024-02-06:
if you have been able to set fleet telemetry configs, you can try using:-
https://developer.tesla.com/docs/fleet-api#fleet_telemetry_errors
and
https://developer.tesla.com/docs/fleet-api#fleet_telemetry_config-get

to troubleshoot the problem

bgk-, 2024-02-12:
The telemetry errors api is exactly what I was hoping for. Apparently our cert wasn't signed... 🤦 

If I encounter any other issues I'll open a new ticket. Thank you!

closed 2024-02-12 by bgk-