teslamotors/fleet-telemetry #109: Question regarding CSR Validation

StephenPorter, 2024-02-06:
We submitted a CSR back in December and are awaiting approval.  Getting telemetry data is critical to the future of our product.  Assuming we've done something incorrectly, we tried to validate the CSR using the new `check_csr.sh` script.

We have our public key registered at our root domain but have issued the CSR with the fleet telemetry endpoint as the common name.

The fleet telemetry endpoint is being hosted on a sub-domain (i.e. `tesla.<rootdomain>.com`), whereas the public key is hosted at the root: `https://<rootdomain>.com/.well-known/appspecific/com.tesla.3p.public-key.pem`

This causes `check_csr.sh` to fail as it reflects on the `CN` from the CSR to try and pull the public key from `https://tesla.<rootdomain>.com`

Does the CSR need to be supplied with the root domain as the common name, or does it need to be supplied with the eventual sub-domain for the fleet telemetry server?

If it turns out the root domain is what is expected, will it still work to have fleet telemetry hosted on a sub-domain?

Bre77, 2024-02-09:
I hope you get a response from Tesla, but my understanding is that it will perform mutual TLS authenticaiton, without checking what domain the vehicle is actually connecting to (like a web browser would). The certificate is to prove who you are, not what domain its connecting to. This means you should be able to instruct the car to connect to an IP address and everything would still work.

StephenPorter, 2024-02-10:
Thanks for the response!  Really appreciate the insight.

For those who are interested, Tesla support verified that we had the incorrect `CN` specified in the request.  The CSR should definitely contain the domain name where your `.well-known` endpoint is hosted.  

Also, there is some sensitivity if you have a discrepancy in white space between your hosted `com.tesla.3p.public-key.pem` and what `openssl` decodes from the CSR.  So make sure to verify with the [check_csr.sh](https://github.com/teslamotors/fleet-telemetry/blob/main/tools/check_csr.sh) script prior to submitting!

closed 2024-02-10 by StephenPorter