teslamotors/fleet-telemetry #114: Fleet Telemetry Ready but Unknown CSR

Bre77, 2024-02-23:
This morning I received the email saying I was Fleet Telemetry ready.

I create a brand new CA cert using my private key, and issued a certificate to my fleet telemetry server. I then used the `fleet_telemetry_config` endpoint to install config on my Model 3. However the `fleet_telemetry_errors` endpoint has told me `certificate signed by unknown authority`

I ran `./check_server_cert.sh conf.json` and it returned:
```
/tmp/tmp.xmvFF6Cnk1: OK
The server certificate is valid.
```

So it would seem my configuration and Fleet Telemetry server are configured correctly, yet my vehicle does not like my certificate authority.

Bre77, 2024-02-23:
I dont think the vehicle is respecting the `port` variable in my config...

patrickdemers6, 2024-02-24:
The vehicle always uses port `443`. The port you specify in `config.json` tells the server where to listen. You just need to make sure public port 443 gets mapped to whatever port you specify.

patrickdemers6, 2024-02-24:
Hopefully at some point the port the vehicle uses will be customizable.

Bre77, 2024-02-24:
> The vehicle always uses port `443`. The port you specify in `config.json` tells the server where to listen. You just need to make sure public port 443 gets mapped to whatever port you specify.

The `check_server_cert.sh` reference, and uses a port configure in the proposed_config.json

There is also a degree of conflict that you can configure the server to use a different port despite the vehicle _not_ being able to. I get it, load balancers, and that's what I'll probably end up doing eventually.

patrickdemers6, 2024-02-24:
That's a great point. When setting up my instance I used 443 to keep it simple so didn't think about that.

StephenPorter, 2024-02-24:
I was messing around with `check_server_cert.sh` using your excellent guide here:
https://dev.to/patrickdemers6/fleet-telemetry-streaming-data-from-your-tesla-2i97

I wasn't sure if it expects the server or client config for verification?  I assumed the client config is probably the payload that you use for the `POST fleet_telemetry_config` endpoint when you are registering a vehicle with fleet-telemetry.  That seems to require a `hostname`, `port` (optional) and `ca` similar to what the script expects.  The way I got the verification script to work was to send the entire certificate chain for my existing server's TLS certificate as a string into the `ca` field of the `config.json` file I passed to the script.

patrickdemers6, 2024-02-24:
The `check_server_cert.sh` script expects the server's configuration file. When creating this `config.json` in tutorial, I instruct to add the CA field (which is cert chain) to the server configuration to make the script work out of the box.

StephenPorter, 2024-02-24:
Thanks!  In that case, do you know if the actual server requires the `hostname` and `ca` fields (i.e. the full certificate chain) to be supplied in it's config file as well?   I hadn't seen those in the docs, but I know things are in flux.  

I think I got thrown off because your tutorial says `path to full certificate chain from previous step` in your example server config file (I interpreted that to be a filename), but `check_server_cert.sh` seemed to just expect a big json string like the client config demands in the `Send configuration to vehicle` section in the tutorial.

patrickdemers6, 2024-02-25:
The server config file doesn't need `hostname` or `ca`. They are both used to tell the script where the web server is to validate certs.

You're right - I'll correct that. Thanks for calling out my oversights!

bassmaster187, 2024-02-26:
> The vehicle always uses port `443`. The port you specify in `config.json` tells the server where to listen. You just need to make sure public port 443 gets mapped to whatever port you specify.

Unfortunately apache is already using port 443 on my server. How did you solve this? Are requests from cars a http request, so I can use anything like ProxyPassMatch / ProxyPassReverse

Bre77, 2024-02-26:
> Unfortunately apache is already using port 443 on my server. How did you solve this? Are requests from cars a http request, so I can use anything like ProxyPassMatch / ProxyPassReverse

Welcome to my problem. You either need a second IP address and bind each application separately, use an SNI proxy infront to do a non decrypting proxy, or like I did, run a second server.

Eventually I plan to rewrite the Fleet Telemetry server in NodeJS and terminate all SSL including the mTLS in Caddy 2.

Bre77, 2024-02-26:
@GaPhi shameless plug, but if you want to test your vehicle with Fleet Telemetry, sign up for a free trial at https://teslemetry.com/ and you can get your fleet telemetry in the browser, over SSE, or web hooks. I suspect if you have an MCU2 or similar, your vehicle will be too old.

GaPhi, 2024-02-26:
I have MCU3 (upgraded thanks to FSD bought).

teslemetry.com does not work : it sees my car but it cannot configure the
fleet-telemetry (my config remains active) and it does not see the
failure...
More, when trying to remove access to my car, it says it is done but
continues to access it... Deleting the account is not achieved even if it
is written it has been done...

Hopefully, Tesla has sent an email with a revocation link that works!

I think this teslemetry.com site is not ready yet to be secure... You are
warned!

Thanks for your help but I will continue to search for a more convincing
solution/diagnostic of my problem.

patrickdemers6, 2024-02-26:
> Unfortunately apache is already using port 443 on my server. How did you solve this? Are requests from cars a http request, so I can use anything like ProxyPassMatch / ProxyPassReverse

@bassmaster187 @Bre77 change rolling out later today to make it so you can set the port of your fleet-telemetry server. Will let you know when it's live.

Bre77, 2024-02-27:
@patrickdemers6 thats awesome, I'll test it as soon as its avaliable, I have a servers in the EU and US ready and waiting :)

@GaPhi if you still have concerns email me at admin@teslemetry.com, but the site did tell you to [revoke with Tesla too](https://auth.tesla.com/en_au/user/revoke/consent?revoke_client_id=ce37f0649b2a-4e4e-8fdc-0f3064b909ff)

patrickdemers6, 2024-02-27:
@Bre77 the change is deployed. I have validated in prod you can now point the car to an alternative port.

Bre77, 2024-02-27:
> @Bre77 the change is deployed. I have validated in prod you can now point the car to an alternative port.

I can confirm my Model 3 is now attempting to connect on the custom port I have specified.

closed 2024-02-27 by patrickdemers6

bassmaster187, 2024-02-27:
> @bassmaster187 @Bre77 change rolling out later today to make it so you can set the port of your fleet-telemetry server. Will let you know when it's live.

I realized this morning that port is now a required field. Such breaking changes should be announced by a mailing list or whatever.