teslamotors/fleet-telemetry #118: best practice (+ maybe a feature request)

Adminius, 2024-02-27:
Hi, what is the best way to implement a logger?

For example while driving these fields are important:
- Location
- Heading
- Battery level (SoC)
- Rated range
- Autopilot state
- Gear
- Speed
- maybe also TPMS
- Temperature inside/outside

But experience with FleeAPI shows, that already 4 fields with 10 seconds interval already exceeding API rate limits.
So, what is the best practice to do it?

Feature Request:
Client(=Logger) asks server for some fields (for example see above)
Server responds only that fields that was changed between last API calls from this client.
In this case Gear, Autopilot state, (speed), temperatures, battery level and some other doesn't change most of the time, so mostly only location and heading will be changed. that reduces traffic but the data quality stays the same.

bassmaster187, 2024-02-27:
10 seconds of Spotify will exceed the traffic of such a logger settings with a 10 seconds interval for hours. So I can't understand why it is limited.

rileymd88, 2024-05-09:
I am also curious how to handle these cases. 

Most loggers today poll the `https://developer.tesla.com/docs/fleet-api#vehicle_data` API every 10-30 seconds to get all the data needed, however it is not possible to get a similar level of detail with fleet telemetry at the moment due to the limits in place.

KarlF-KFLOE, 2024-05-09:
My understanding is that 10 fields per minute is a temporary constraint and subscription tiers offering more fields per minute will be available by the end of Q2 (fast approaching!)  I'm hoping this is still the timeline (and that the cost is reasonable) as this is critical for our own use case!

kiler129, 2024-05-14:
Is the limit still in place in reality? The [current Tesla Fleet API docs](https://developer.tesla.com/docs/tesla-fleet-api#fleet-telemetry) state the following:

[image]

The same section [had the limit listed in 12/2023](https://web.archive.org/web/20231225131305/https://developer.tesla.com/docs/fleet-api#fleet-telemetry):
[image]

The section was [tweaked in 01/2024](https://web.archive.org/web/20240105100843/https://developer.tesla.com/docs/fleet-api#fleet-telemetry):
[image]

Throughout January, February, and until [somewhere in March](https://web.archive.org/web/20240330004144/https://developer.tesla.com/docs/fleet-api#fleet-telemetry) the section listed the limit. In March the "Fleet Telemetry" section was revised to the currently-online abbreviated version.

bassmaster187, 2024-05-14:
Limits are still active:

{
	"response": null,
	"error": "Total data transfer rate is too high. Please reduce field count or increase interval_seconds for some fields.",
	"error_description": "",
	"txid": "xxxxx"
}

patrickdemers6, 2025-01-22:
Limits have been removed for some time now. You should be able to implement your desired functionality much easier now.

closed 2025-01-22 by patrickdemers6