teslamotors/fleet-telemetry #119: Fleet Telemetry doesn't work (anymore)

bassmaster187, 2024-02-27:
I received yesterday messages, but today it doesn't seem to get any messages. Maybe it is related to the port feature:
https://github.com/teslamotors/fleet-telemetry/issues/114#issuecomment-1965515299

https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/fleet_telemetry_config
with port set to 4445 will response:

```
{
	"response": {
		"updated_vehicles": 1
	}
}
```

Car is online.

https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/partner_accounts/fleet_telemetry_errors
just old errors from yesterday. 

Docker up and running
```
2024/02/27 11:44:09 maxprocs: Leaving GOMAXPROCS=4: CPU quota undefined
time="2024-02-27T11:44:09Z" level=info msg=starting
time="2024-02-27T11:44:09Z" level=info msg=status_server_configured
time="2024-02-27T11:44:09Z" level=info msg=profiler_started
time="2024-02-27T11:44:09Z" level=info msg="registered kafka for namespace: tesla_telemetry"

```

http://localhost:8080/status
> ok
BTW: it would be nice to see some more infos like connected cars...

if i connect from a browser i can see a connection has been declined:
```
2024/02/27 12:09:40 http: TLS handshake error from 78.43.56.144:26896: EOF
2024/02/27 12:09:40 http: TLS handshake error from 78.43.56.144:26897: EOF
2024/02/27 12:09:41 http: TLS handshake error from 78.43.56.144:26898: tls: client didn't provide a certificate

```

What else can I check?

GaPhi, 2024-02-27:
> #114 (comment)

I have the same problem : no error from partner account fleet telemetry
using port 443 (via HAProxy thanks to the SNI which is working according to
the logs).

HAProxy : 142.93.131.240:44830 [27/Feb/2024:12:48:56.458] incoming_443
tesla_telemetry_gaphi_net_backend_tcp/tesla_telemetry_server 2/1/109 2753
-- 3/3/0/0/0 0/0 {tesla-telemetry.gaphi.net} SNI:tesla-telemetry.gaphi.net
Telemetry : 2024/02/27 12:48:56 http: TLS handshake error from
192.168.1.34:47442: *tls: client didn't provide a certificate*

PartnerAPI : {"response":{"fleet_telemetry_errors":[]}}

Now I am trying with port 444 (direct NAT to telemetry) :

TeslaAPI getConfig : {"response":{"synced":true,"config":{"hostname":"
tesla-telemetry.gaphi.net
","ca":"...","exp":1740488077,"port":444,"fields":{"BmsFullchargecomplete":{"interval_seconds":60},"ChargePort":{"interval_seconds":30},"DriveRail":{"interval_seconds":60},"Gear":{"interval_seconds":60},"Soc":{"interval_seconds":30}},"alert_types":["service"]}}}

Do you have a way to trigger the client (vehicle) connection to the server
? It takes so much time when trying a new configuration waiting for a
connection for several hours or days...
I tried to register the ChargePort event and open/close it with tesla-api,
but it seems to not be enough...

Thanks

GaPhi, 2024-02-27:
I have now new messages from fleet-telemetry server :
2024/02/27 13:29:03 http: TLS handshake error from 156.28.6.11:60557: EOF
2024/02/27 13:29:03 http: TLS handshake error from 156.28.6.11:22421: *tls:
peer doesn't support any of the certificate's signature algorithms*

My signature is using :
    Signature Algorithm: *sha512WithRSAEncryption*
Probably one of the problems then.

I try with another simpler signature algorithm :
    Signature Algorithm: *sha256WithRSAEncryption*

But key is still 4096 bits, maybe too much too :
            Public Key Algorithm: *rsaEncryption*
                Public-Key: (*4096 bit*)
Waiting for next message...

It would be great to have documented vehicle limitations...

bassmaster187, 2024-02-28:
> I have now new messages from fleet-telemetry server : 2024/02/27 13:29:03 http: TLS handshake error from 156.28.6.11:60557: EOF 2024/02/27 13:29:03 http: TLS handshake error from 156.28.6.11:22421: *tls:

Your Name and IP Address seems to be from France / Europe. I'm from Germany, so we are both on the Europe Server. 
I double check my settings twice with @Bre77 but my server don't work and his works. But the difference seems to be his region is USA. Maybe thats the problem?

bassmaster187, 2024-02-28:
@patrickdemers6 could you please check if your changes of the port settings from issue #114 doesn't affect region Europe.

patrickdemers6, 2024-02-28:
I just configured streaming on an EU vehicle using the EU endpoint. No change to the config I sent to `/api/1/vehicles/fleet_telemetry_config`.

Could you email me a VIN which is failing, client id, and any other pertinent details? fleetapisupport@tesla.com

bassmaster187, 2024-02-29:
the port of my fleet_telemetry_config was send as a string instead of integer. 
@patrickdemers6 Thank you very much.

@GaPhi maybe you have the same problem?

closed 2024-02-29 by bassmaster187

GaPhi, 2024-02-29:
No, I did send the port as an integer... Thank you for the idea.

This is probably more a car compatibility I suppose as all API calls
succeed (including synced=true after a while) but no telemetry connection
is established, whatever the certificates I provide.

I tried :
- signed by my private CA or a public CA (set for check_server script and
config api call),
- same keys as the Tesla partner CSR registration or dedicated ones with
different algorithms,
- using shared port 443 (thanks to the client SNI) or direct with port 444
without SNI usage,
- each time, the check_server script is OK.

I do not have any other ideas right now...
Maybe a Tesla support can help, I will try to contact them ; with the VIN
they could probably quickly answer the question of the car capability. But
I would hope an error message during config call or at least the partner
fleet error call if the car is not capable.