teslamotors/fleet-telemetry #121: Support older Model S/X

bassmaster187, 2024-03-01:
Is it on the roadmap to support older Model S/X? 

if not, is it because of missing virtual key (bluetooth) on older Model S/X?

Adminius, 2024-03-01:
And if not, how to deal with it?

jbanyer, 2024-03-02:
We support older vehicles via the Fleet API (not bluetooth) by detecting that they don't support command signing, and using the regional API endpoint directly instead of the command proxy.

Bre77, 2024-03-02:
> We support older vehicles via the Fleet API (not bluetooth) by detecting that they don't support command signing, and using the regional API endpoint directly instead of the command proxy.
> 
> I can't comment on bluetooth.

This is Fleet Telemetry specific @jbanyer, my understandingis the hardware in older vehicles does not support Fleet Telemetry at all.

aaronpkahn, 2024-03-14:
There is no plan to support first generation S/X computers, but we may support second gen this year.

Leftwitch, 2024-09-25:
> There is no plan to support first generation S/X computers, but we may support second gen this year.

Any news on this topic ? If not with fleet telemetry how to handle older Models then, since I cant poll vehicle_data due to rate limits aswell, I don't quite know how to handle this right now, thats why I hope you got some news 6 Months later 😢

ThomasAlxDmy, 2024-11-28:
Leaving this open until S/X gen 1 & gen 2 are supported.