teslamotors/fleet-telemetry #123: Which field provides the ChargingState?

virtualzone, 2024-03-02:
Hi there,

I'm trying to retrieve the ```ChargingState``` from the telemetry data. I assumed this was returned in the ```ChargeState```.

I assumed that ```ChargeState``` had type ```charging_value```, so I could check i.e. for ```ChargeStateDisconnected``` (enum value 1).

However, ```ChargeState``` only returns a ```string_value``` which is ```Idle```whenever the vehicle is *not* charging. 

Help is appreciated, thanks in advance!

jbanyer, 2024-03-02:
I'm also interested in this question. Our use case requires knowing when the vehicle gets plugged in or unplugged.

purcell-lab, 2024-03-15:
ChargeState:/ ChargeEnableReuest: seems to change when charging starts/ stops:

https://github.com/teslamotors/fleet-telemetry/issues/127#issuecomment-1999095245

peetermeos, 2024-03-24:
As far as I can see, ChargingState returns rather useless Idle/Enabled string value, the expected enum ChargingState is always default (ie. ChargingStateUnknown).  So in short, it does not seem to be yet fully implemented. Emailed support about it, no response yet.

agbpatro, 2024-04-11:
For [ChargingState](https://github.com/teslamotors/fleet-telemetry/blob/0431c0fa97158bbac5d2c7fb2eb0b517e2c95530/protos/vehicle_data.proto#L12) , value should be stored in [string](https://github.com/teslamotors/fleet-telemetry/blob/0431c0fa97158bbac5d2c7fb2eb0b517e2c95530/protos/vehicle_data.proto#L213)

stx, 2024-04-11:
@agbpatro Developers want to know how to access the ChargingState values like Disconnected, Charging, Complete, listed here: https://github.com/teslamotors/fleet-telemetry/blob/0431c0fa97158bbac5d2c7fb2eb0b517e2c95530/protos/vehicle_data.proto#L181

ChargeState does not provide this (it provides "Idle" or "Enable") and it appears there is no way to retrieve this field. This is a very important field for virtually all third party apps. Is there a way to get the ChargingState?

agbpatro, 2024-04-11:
I see yeah. That is not supported yet. Will update the thread here once its available

jbanyer, 2024-04-11:
@agbpatro thanks, just reiterating @stx's comment that this issue is preventing us from moving to fleet telemetry. I did email fleetapisupport@tesla.com about it on 18th March but haven't had a reply. Cheers.

Bre77, 2024-04-14:
@jbanyer FYI we have observed three states from Streaming that map to polled values:

"Idle": "stopped", #streaming
"QualifyLineConfig": "starting", #streaming
"Enable": "charging", #streaming

https://github.com/Teslemetry/hass-teslemetry/blob/eb3e886e8222010d2e131c5092e5dd94f0444a18/custom_components/teslemetry/sensor.py#L49

Urkman, 2024-06-19:
> I see yeah. That is not supported yet. Will update the thread here once its available

Any Updates on this?

And is there currently a way to get the information that the user connects or disconnects the charger?
And when the charging completes (not stopped)?

Urkman, 2024-06-26:
One more on this:
A user off my App was charging at a "32 kWh am EnBW HPC".
But the `charging_state` was `Idle` for the whole time...

Any idea on this?

jbanyer, 2024-12-10:
@virtualzone  I think this can be closed now, given the `DetailedChargeState` field.