teslamotors/fleet-telemetry #126: fleet_telemetry_config error 404 vin not_found

jbanyer, 2024-03-10:
Fleet telemetry was working for me, but now I'm receiving a 404 vin not found when trying to (re)configure it on a vehicle which was working before.

endpoint: `/api/1/vehicles/fleet_telemetry_config`
reply: http 404
```
{
  response: null,
  error: '<REDACTED VIN> not_found',
  error_description: ''
}
```

agbpatro, 2024-03-11:
are you sure you still have access to that VIN from your account?

jbanyer, 2024-03-11:
Discovered that this error is happening because I'm trying to configure fleet telemetry using a Tesla account with DRIVER (non-owner) access to the vehicle.

Per #117 it would be great if drivers could enable fleet telemetry on a vehicle.

One factor in my confusion is #116 , which meant I could delete the fleet telemetry config using the driver account but could not reconfigure it.

Some suggestions for the devs:
- the error message could be improved eg "unable to configure fleet telemetry with driver access". The current message is misleading because a driver account *does* have visibility of the vin
- the [docs](https://developer.tesla.com/docs/fleet-api#fleet_telemetry_config-create) could be improved to clarify that this endpoint is only available to accounts with owner access

aaronpkahn, 2024-03-14:
@jbanyer DRIVER access will be supported soon.  We'll publish the change to the [announcements](https://developer.tesla.com/docs/fleet-api#announcements-amp-api-changelog) section

closed 2024-03-14 by aaronpkahn

amirhmk, 2024-04-01:
@jbanyer how did you end up solving this? I am using my own vehicle for testing, and I had previously setup 3rd party, so I assumed fleet telemetry would be setup already. 

> Ensure your virtual key has been added to the vehicle you intend to configure. To add your virtual key to the vehicle, redirect the owner to https://tesla.com/_ak/your-domain.com. **If using authorization code flow, the owner of the vehicle must have [authorized your application](https://developer.tesla.com/docs/fleet-api#third-party-token) with vehicle_device_data scope before they are able to add your key.**

Either way, I also added my virtual key using the first approach key, and I was able to confirm that the key was added to my vehicle by checking the vehicle settings and the iOS app notification. However, I'm still getting the same error. Any ideas?

jbanyer, 2024-04-02:
> Any way, I also added my virtual key using the first approach key, and I was able to confirm that the key was added to my vehicle by checking the vehicle settings and the iOS app notification. However, I'm still getting the same error that my VIN is `not_found`. Any ideas?

@amirhmk are you attempting to configure fleet telemetry using a Tesla account which is the OWNER of the vehicle? My problem (and this ticket) was because I was trying to use a Tesla account with DRIVER access to configure fleet telemetry, which doesn't work currently.

amirhmk, 2024-04-02:
Yes @jbanyer, I am using my own vehicle and account to get the fleet telemetry to work, but getting the exact same error as you. Now I'm wondering if my whole TLS / CSR setup is correct or even matters at this stage since I'm getting the same error if I change the config in some way.