teslamotors/fleet-telemetry #127: Which fields provide charging current and voltage?

jbanyer, 2024-03-12:
Does the fleet telemetry provide actual charging current and voltage?

The fields in Fleet API are `charger_actual_current` and `charger_voltage`.

Bre77, 2024-03-12:
The following would be of use to you @jbanyer 

ACChargingEnergyIn
ACChargingPower
ChargeCurrentRequest
ChargeCurrentRequestMax

My Teslemetry community started tracking what's working for which vehicles and what the values today:
https://docs.google.com/spreadsheets/d/1TRGLpBU2cPWBC1TZam5xx2n-kSSwcE3p95LBVtMihLc/edit?usp=sharing

jbanyer, 2024-03-13:
Thanks @Bre77. I've made some contributions to your Google sheet.

I've looked and tested fairly carefully and I can not find any fields for the [actual] charging amps or voltage.

If anyone from Tesla can comment that would be great. As a charging app [chargehq.net], amps and voltage are fairly critical.

purcell-lab, 2024-03-15:
I haven't seen a voltage (apart from battery pack voltages which looks interesting), but I do see charging current in various forms:

ChargeCurrentRequest seems to be what has been called for.
ChargeAmps seems to be what is currently being delivered.
ChargeCurrentRequestMax as per the name.

ACChargingPower is returned as a float, unlike the API where it is returned as an int.
EnergyRemaining seems to be battery soc in kWh, which is useful. 
ChargeState:/ ChargeEnableReuest: seems to change when charging starts/ stops

```
data: {"data":{"ChargeCurrentRequest":"16"},"createdAt":"2024-03-15T07:22:54.429487543Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"Soc":"80.913"},"createdAt":"2024-03-15T07:23:04.429417483Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"ChargerPhases":"2","ChargeAmps":"16","ChargeLimitSoc":"100"},"createdAt":"2024-03-15T07:23:24.429636642Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"ChargeState":"Enable","Soc":"81.670","ChargeLimitSoc":"100","ChargeEnableRequest":"true","ChargeCurrentRequestMax":"16","ChargerPhases":"2","ChargeAmps":"5"},"createdAt":"2024-03-15T07:28:44.466097926Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"ACChargingPower":"3.700"},"createdAt":"2024-03-15T07:28:44.466332093Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"ChargeCurrentRequest":"5"},"createdAt":"2024-03-15T07:29:14.466225601Z","vin":"5YJ3F7EC5LF698201
data: {"data":{"EnergyRemaining":"57.240"},"createdAt":"2024-03-15T07:31:04.466315178Z","vin":"5YJ3F7EC5LF698201"}

data: {"data":{"ChargeState":"Idle","Soc":"81.851","ChargeLimitSoc":"100","ChargeEnableRequest":"false","ChargeCurrentRequestMax":"16","ChargerPhases":"2","ChargeAmps":"5"},"createdAt":"2024-03-15T07:34:44.465784758Z","vin":"5YJ3F7EC5LF698201"}
```

jbanyer, 2024-04-11:
@agbpatro any chance you could comment on this one? Thanks.

krishR007, 2024-06-27:
> I haven't seen a voltage (apart from battery pack voltages which looks interesting), but I do see charging current in various forms:
> 
> ChargeCurrentRequest seems to be what has been called for. ChargeAmps seems to be what is currently being delivered. ChargeCurrentRequestMax as per the name.
> 
> ACChargingPower is returned as a float, unlike the API where it is returned as an int. EnergyRemaining seems to be battery soc in kWh, which is useful. ChargeState:/ ChargeEnableRequest: seems to change when charging starts/ stops
> 
> ```
> data: {"data":{"ChargeCurrentRequest":"16"},"createdAt":"2024-03-15T07:22:54.429487543Z","vin":"5YJ3F7EC5LF698201"}
> data: {"data":{"Soc":"80.913"},"createdAt":"2024-03-15T07:23:04.429417483Z","vin":"5YJ3F7EC5LF698201"}
> data: {"data":{"ChargerPhases":"2","ChargeAmps":"16","ChargeLimitSoc":"100"},"createdAt":"2024-03-15T07:23:24.429636642Z","vin":"5YJ3F7EC5LF698201"}
> data: {"data":{"ChargeState":"Enable","Soc":"81.670","ChargeLimitSoc":"100","ChargeEnableRequest":"true","ChargeCurrentRequestMax":"16","ChargerPhases":"2","ChargeAmps":"5"},"createdAt":"2024-03-15T07:28:44.466097926Z","vin":"5YJ3F7EC5LF698201"}
> data: {"data":{"ACChargingPower":"3.700"},"createdAt":"2024-03-15T07:28:44.466332093Z","vin":"5YJ3F7EC5LF698201"}
> data: {"data":{"ChargeCurrentRequest":"5"},"createdAt":"2024-03-15T07:29:14.466225601Z","vin":"5YJ3F7EC5LF698201
> data: {"data":{"EnergyRemaining":"57.240"},"createdAt":"2024-03-15T07:31:04.466315178Z","vin":"5YJ3F7EC5LF698201"}
> 
> data: {"data":{"ChargeState":"Idle","Soc":"81.851","ChargeLimitSoc":"100","ChargeEnableRequest":"false","ChargeCurrentRequestMax":"16","ChargerPhases":"2","ChargeAmps":"5"},"createdAt":"2024-03-15T07:34:44.465784758Z","vin":"5YJ3F7EC5LF698201"}
> ```

Can you explain how i connect telemetry with my server ?

jbanyer, 2024-12-02:
Recent investigation suggests that:
- `ChargeAmps` is the actual charge amps (same as `charger_actual_current` in the polling API)
- `ChargeCurrentRequest` is the amps as set by `set_charging_amps` or in the app
- `ChargeCurrentRequestMax` is the max amps offered by the EVSE

I still haven't found a field for AC voltage...

patrickdemers6, 2025-01-16:
`charger_voltage` is returned by `ChargerVoltage`.
`charger_actual_current`  is returned by `ChargeAmps`.

Sorry it took so long to make charger voltage available.

closed 2025-01-16 by patrickdemers6

EduardoLaranjo, 2025-03-20:
Hi @patrickdemers6, which firmware version was the charger_voltage field released in