teslamotors/fleet-telemetry #128: Fleet Telemetry Server is started but no data is received

PrriyaR, 2024-03-13:
Moved this out of https://github.com/teslamotors/fleet-telemetry/issues/122 and created new issue

@patrickdemers6 Update: I am able to fix the certificate issue. And using acme Bash https://github.com/acmesh-official/acme.sh/wiki/Issue-a-cert-from-existing-CSR i generated the certificates.

And now the server is running. Where do i check the logs?
Here is what i got when docker compose up. I have already created telemetry config for my tesla vehicle with full chain cert. No telemetry data is received to the server

$ docker-compose up
[+] Running 1/0
 ✔ Container fleetfiles-app-1  Created                                                                                                                                                     0.0s
Attaching to app-1
app-1  | 2024/03/13 19:35:12 maxprocs: Leaving GOMAXPROCS=1: CPU quota undefined
app-1  | time="2024-03-13T19:35:12Z" level=info msg=config_skipping_empty_metrics_provider
app-1  | time="2024-03-13T19:35:12Z" level=info msg=starting
And no telemetry errors from fleet_telemetry_errors endpoint

{
"response": {
"fleet_telemetry_errors": []
}
}

Still check_server_cert.sh fails for the cert that i got from acme client.

```
/tmp/tmp.FMNmhcx9pU: CN = <domain name>
error 20 at 0 depth lookup:unable to get local issuer certificate
/tmp/tmp.FMNmhcx9pU: CN = <domain name>
error 20 at 0 depth lookup:unable to get local issuer certificate
The server certificate is invalid.
```

patrickdemers6, 2024-03-14:
Are you including `ca` in the `config.json` file you're passing to check_server_cert.sh?

patrickdemers6, 2024-03-14:
Yeah you need to make sure the TLS connection is being terminated at the fleet-telemetry server, not a load balancer or something else sitting out front. 

Are you having any logs spit out once the server is started (like connection attempts with bad certs)? If not, it may be a sign the fleet-telemetry server isn't accessible at all to the outside world.

closed 2025-01-22 by patrickdemers6