teslamotors/fleet-telemetry #130: Config Sync - Questions/Observations

KarlF-KFLOE, 2024-03-15:
Thank you for all the hard work making Fleet Telemetry available to the developer community. We have successfully switched our app from polling to fleet telemetry and it is working well.
 
One thing we have noticed is that it takes a while for a vehicle to start sending data after it has been enrolled via 'fleet_telemetry_config create'.  For our test fleet it has taken anywhere from a few hours to almost a full day for the sync to happen.

- What determines the timing of the vehicle connecting to the backend to receive the target config?
- Is there any way to force the sync or make the timing more predictable?  If not today, are there any plans to improve this in the future?
- Should telemetry data start as soon as the sync parameter returns true via 'fleet_telemetry_config get'?

One of our test vehicles is a 2018 Model S (Gen 2).  We know this vehicle is not (yet) supported by Fleet Telemetry but we are successfully able to push a config to it (sync flips to true) but we do not receive telemetry (as expected).  Should an error be returned when targeting an unsupported model?

Any insight you can offer is much appreciated.  Keep up the great work!

patrickdemers6, 2024-03-20:
Hi @KarlF-KFLOE thanks for reaching out. Glad you're having some success with streaming.

A future vehicle firmware release should help reduce delay before data is sent. Out of curiosity, do you have long intervals specified? To my knowledge there's no way to force the config to sync immediately, though I've never seen it take more than a minute or two.

Happy to help investigate what took so long to see if there's an underlying issue. Feel free to email client ID/vin/timestamp of the slow sync to fleetapisupport@tesla.com.

You're right, setting config on a vehicle that doesn't support streaming is a bit odd. I'll keep you posted on this.

KarlF-KFLOE, 2024-03-25:
Thanks @patrickdemers6 
 
After more testing, this seems isolated to the "synched" flag and not the configuration push itself. When we push a config change to a vehicle the telemetry stream reflects the change quickly (minutes) but "synched" shows false for many hours. 

Interestingly, our '22 MXP (FW 2023.44.30.25) consistently takes almost a day for "synched" to show "true".  Our '23 MYP (FW 2024.2.7) flips more quickly, but still takes many hours.
 
Our use case requires frequent telemetry, so we don't have any long intervals specified.  We're currently receiving "Location" every 7 seconds and we're eagerly awaiting the ability to receive more fields/minute! :)

If you want to dig deeper, I'm happy to test again and email the info requested.

agbpatro, 2024-03-25:
> Thanks @patrickdemers6
>
> After more testing, this seems isolated to the "synched" flag and not the configuration push itself. When we push a config change to a vehicle the telemetry stream reflects [...]

we have identified the issue for the delay for reporting on synced as `false` and are working on fixing it

KarlF-KFLOE, 2024-03-25:
Brilliant.  If you'd like us to test this (or any other issues) when the fix rolls out, we're happy to help!

agbpatro, 2024-03-26:
> Brilliant. If you'd like us to test this (or any other issues) when the fix rolls out, we're happy to help!

Thanks for raising the issue. It should have been mitigated by now.

KarlF-KFLOE, 2024-03-28:
@stx Agreed.  It's still taking a prolonged period (>24 hours) for the "synced" flag to flip, despite the telemetry stream quickly reflecting the configuration change.  I'll send examples to fleetapisupport@tesla.com

agbpatro, 2024-03-28:
Yes please send examples to fleetapisupport@tesla.com and we will take a look

KarlF-KFLOE, 2024-04-04:
This is now resolved for us.  Pushing a new config and seeing "synced" flip from "false" to "true" now takes less than a minute.  Thanks for the fix!  Nice job :)

closed 2024-04-04 by KarlF-KFLOE