teslamotors/fleet-telemetry #133: Confusion about Server Setup

amirhmk, 2024-03-21:
Hi thre, 

We have succesfully started using the 3rd party API, and it's working well. Now we have a need to start streaming data, so we need to setup fleet-telemetry as well. 

I have some prior experience with K8s, but I'm finding the steps to setup the server a little confusing. If I want to try out setting this whole service up using a minikube cluster to make sure everything is sorted, how do I go about doing that? Is that possibe at all? I have installed the helmcharts, but I'm getting a `http://10.244.0.3:8080/status 10.244.0.3:8080: connect: connection refused` error from the dashboard. 

I see that the README was updated to include more details on the previous steps, but I'd appreciate if you can please clarify/add more details to the actual server part. 

For example, what is the `docker-compose.yml` file for? Is it possibel to test this locally at all, or we'd have to setup all the cloud infra first using K8s? Would appreciate some pointers!!

amirhmk, 2024-03-26:
Still confused about the server setup, @patrickdemers6 perhaps we could get a diagram to visualize the system? Any pointers would be great :)

patrickdemers6, 2024-03-27:
Hey @amirhmk, I'll be honest I'm not familiar with helmcharts. Maybe @aaronpkahn can help with that part?

You can't test a server locally since vehicles need a publicly accessible endpoint to access.

Personally, I would avoid using K8s initially as it adds complexity. To start simple, I'd recommend using Docker or even running the binary directly.

I have an article I wrote which describes steps for Docker: https://dev.to/patrickdemers6/fleet-telemetry-streaming-data-from-your-tesla-2i97. For running the binary, you'd have to build the go application. Then, run it and pass in `-config=<path_to_config>`.

Things to look out for:
- The port the server attaches to needs to bind to the public network interface
- Ensure mTLS connections are terminated at the server

> For example, what is the docker-compose.yml file for?
Not sure if you're asking docker-compose.yml in general or as a part of helm charts?

Happy to answer more specific questions you have.

amirhmk, 2024-04-01:
Great, thank you for the article above @patrickdemers6 , I think adding this to the README would help a lot of people!!

So I have setup all the steps, and was able to get `The server certificate is valid` message when trying out the tool you've provided. 

However, when I now try to send the telemetry config to the vehicles, I get a 404 with an `MY_VIN_ID not_found` error. Now I want to make sure my setup is correct. 

For our 3rd party APIs, we're using `tesla.domain.com` and have registered the CSR using the `/register` endpoint and that's working well. 

For the Fleet Telemetry application, we're using `tesla-fleet.domain.com` (which I checked ther cerfificates above and it works), however, I'm not sure if we need to also call the `/register` endpoint in this case? Can we even have 2 different domains?

patrickdemers6, 2024-04-03:
@amirhmk  Could you send an email to fleetapisupport@tesla.com? Include you client ID and VIN. Once we come to a conclusion, we can post the result here for the community.

amirhmk, 2024-04-03:
Great point @patrickdemers6, I was told the following in the email:

> Since you are not onboarded for T4B, all API calls interacting with your own vehicles need to use a third party token. https://developer.tesla.com/docs/fleet-api#third-party-token

I was then told to contact Tesla Fleet by visiting https://www.tesla.com/fleet, however, from first glance I was under the impression that one does not necessarily need to participate in the Tesla 4 Business program. There seems to be a discrepancy from this sentence from this repo's README

> Owners can allow registered applications to receive telemetry securely and directly from their vehicles. **This reference implementation can be used by individual owners as is or by fleet operators who can extend it to aggregate data accross their fleet.**

Could you please clarify?

Bre77, 2024-04-03:
> Great point @patrickdemers6, I was told the following in the email:
> 
> > Since you are not onboarded for T4B, all API calls interacting with your own vehicles need to use a third party token. https://developer.tesla.com/docs/fleet-api#third-party-token
> 

To me it sounds like you have created a partner token, but not authorised your own Tesla account to get a third party token.

Is your own application listed at https://accounts.tesla.com/account-settings/security?tab=tpty-apps

If not, you have missed the third party token step from the documentation, that they linked you in the email.

amirhmk, 2024-04-04:
Yes, exactly @jordan-bonecutter, we have my company's app registered under `developer.tesla.com` and I am using my personal vehicle to do dev work. I had granted permission to my vehicle, but I thought the fleet_telemetry_config would use the `partner_token` and not the 3rd party token. 

Anyways, I am waiting to see some logs on the fleet-telemetry server now to complete the loop. So far nothing (not even errors).

Bre77, 2024-04-04:
> So far nothing (not even errors). 

Did you check the fleet telemetry errors API? https://developer.tesla.com/docs/fleet-api#partner-endpoints

amirhmk, 2024-04-04:
> > So far nothing (not even errors).
> 
> Did you check the fleet telemetry errors API? https://developer.tesla.com/docs/fleet-api#partner-endpoints

Yep, I used that but there's nothing. I tried emailing now to see if there is anything that I'm not catching. 

I do see a log like this on the server, but I was getting these sort of requests before as well:

```
2024/04/04 02:43:10 http: TLS handshake error from xxxxxx:59065: read tcp xxxxx:443->xxxxxx:59065: read: connection reset by peer
2024/04/04 03:02:38 http: TLS handshake error from xxxxx:59048: EOF
```

amirhmk, 2024-04-18:
Okay so I finally started getting data now. I wanted to share some of the details that made the setup work. 

1. I wasn't getting any error logs from `fleet_telemetry_errors` or on the server initially. After following up with the tesla fleet email, they wanted my confirmation that I have in fact requested a new CSR (I'm not sure if this is gonna be the case for everyone if their first CSR is correct). 
2. After confirming this, I had to make sure to use the **3rd party token (OWNER)** to send the configs to the vehicle using `POST /fleet_telemetry_config`. 
3. After seeing `Client sent an HTTP request to an HTTPS server.` errors and 502, I noticed that I was using an Application Load Balancer (ALB) listening on HTTPS 443, but I had to use a **Network Load Balancer (NLB) on TCP 443**. AWS's passthrough mTLS for ALB made me think it meets the requirements.

amirhmk, 2024-04-18:
@Bre77 Okay I'm facing issues again, and I think I know why, but I'm not sure what the best way to set this up would be. Quick question, I can see in the tutorial above that we can use two different domains, one for server and one for public use. 

Currently my public key is hosted at the same URL as my fleet server. `telemetry-server.domain.com/.well-known/appspecific/com.tesla.3p.public-key.pem`. However, I need to move it to a different subdomain as it's conflicting with LB settings. Before I had it on `tesla.domain.com` but when I submitted a CSR for telemetry-server it seems to have replaced the original `tesla.domain.com` one. 

What's the correct setup here?

Bre77, 2024-04-18:
@amirhmk whats your "partner domain" on https://developer.tesla.com because I think your making this difficult for yourself.

This is an example of a working implementation:

- Partner Domain: teslemetry.com
- Public Key: teslemetry.com/.well-known/appspecific/com.tesla.3p.public-key.pem
- API call made to register teslemetry.com (https://developer.tesla.com/docs/fleet-api#register)
- CSR was created for teslemetry.com
- Fleet Telemetry server: na-west.teslemetry.com & eu-west.teslemetry.com each with their own SSL cert signed by the private key

Your partner domain should probably be domain.com, but if you want it to be tesla.domain.com, then thats what you need to set in the portal, register with the API, and create a CSR for. I actually believe that the domain of your Fleet Telemetry server is entirely irrelevant to all of this, because there is no common name checks, only certificate chain checks.

amirhmk, 2024-04-22:
Thanks for your message @Bre77. Indeed it does seem like I have complicated this setup a little bit. 

I started separating these out now, and sent new configs, but I'm getting an error again. 

So now my setups looks like:

1. Partner domain: `tesla-fleet.matt3r.ai` (public key hosted at `https://tesla-fleet.matt3r.ai/.well-known/appspecific/com.tesla.3p.public-key.pem`
2. Fleet Telemetry server: `tesla-server.matt3r.ai` with a cert and keys created accordingly and passed when starting up the server. 
3. Both `check_csr.sh` and `check_server_certs.sh` pass.
4. When calling the errors endpoint, I see something like this:
 
```
webconnection error: x509: certificate is valid for tesla-fleet.matt3r.ai, not tesla-server.matt3r.ai\" cm_type=stream",
                "error_name": "cloud_manager_error",
                "hostname": "tesla-server.matt3r.ai",
```

Which makes me thing we do need to let Tesla know about the Telemetry Server hostname, as it's something we have to wait on Tesla from this in the README:

> Wait for Tesla to process your CSR. This may take up to two weeks. Once complete, you will receive an email from Tesla. The generated certificate will not be sent back to you; it is attached to your account on the backend and is used internally when configuring a vehicle to stream data.

@patrickdemers6 Can you confirm this please?

patrickdemers6, 2024-04-22:
Is the server certificate generated for `tesla-server.matt3r.ai`? The error indicates the car is trying to connect to `tesla-server.matt3r.ai` but the SSL cert the server is using is for `tesla-fleet.matt3r.ai`

amirhmk, 2024-04-22:
> Is the server certificate generated for `tesla-server.matt3r.ai`? The error indicates the car is trying to connect to `tesla-server.matt3r.ai` but the SSL cert the server is using is for `tesla-fleet.matt3r.ai`

Yes the certificate is set correctly I believe, I verified using the `./chek_server_cert`. After looking at the logs again though, I noticed the following line:

> url of the fleet-telemetry server, **must match the root domain (second-level + first-level domain) of the registered application**

I have kept `tesla-fleet.matt3r.ai` as is, and have switched the hostname for the server to be `server.tesla-fleet.matt3r.ai` now to match above. Still getting a similar error:

```
webconnection error: x509: certificate is valid for tesla-fleet.matt3r.ai, not server.tesla-fleet.matt3r.ai\" cm_type=stream"
```

Bre77, 2024-04-22:
I went to your fleet Telemetry server and it has the wrong common name, so you need to recreate your Fleet Telemetry server cert to match its domain.

Turns out Tesla is doing common name checking.

amirhmk, 2024-04-22:
> I went to your fleet Telemetry server and it has the wrong common name, so you need to recreate your Fleet Telemetry server cert to match its domain.
> 
> Turns out Tesla is doing common name checking.

Doesn't that mean server.tesla.domain.com (Server) and tesla.domain.com (partner domain, public key, etc.) would work?

Bre77, 2024-04-22:
What port is your fleet Telemetry server on, because https://server.tesla-fleet.matt3r.ai has a cert for tesla-fleet.matt3r.ai

amirhmk, 2024-04-22:
Okay it's fully fixed now. When updating the DNS records I pointed server.tesla-fleet to the wrong ip, resulting in cert errors. 

In hindsight, I know that the `fleet_telemetry_config create` endpoint does state `config.hostname` needs to be a subdomain of the `partner_domain`, it'd be good for it to explicitly check this and throw an error if not. 

Thanks!!

closed 2024-04-25 by patrickdemers6

ngleader, 2025-02-22:
@Bre77 @amirhmk 
I would really appreciate your help.
 
my domains
- app domain : app.mydomain.com => registered   POST /api/1/partner_accounts
- public key hosted: app.mydomain.com/.well-known/appspecific/com.tesla.3p.public-key.pem
- fleet telemetery domain : telemetery.mydomain.com

and how can I make server_cert, server_key, ca ?
I have attempted multiple times, but I keep encountering issues, so I am seeking your guidance

Bre77, 2025-02-22:
> @Bre77 @amirhmk
> I would really appreciate your help.
>
> my domains
> - app domain : app.mydomain.com => registered   POST /api/1/partner_accounts
> - public key hosted: [...]

I don't think think your origin domains can be app.domain and your telemetry be telemetry.domain, the telemetry would need to be telemetry.app.domain