teslamotors/fleet-telemetry #141: HTTP server panics

slashmili, 2024-04-04:
I'm running docker image `tesla/fleet-telemetry:v0.1.13` 

After adding the vehicle config via `api/1/vehicles/:id/fleet_telemetry_config` endpoint. I see this error in the `fleet-telemetry` logs

```
{"activity":true,"context":"fleet-telemetry","level":"info","method":"GET","msg":"request_start","remote_ip":"192.168.130.103:12842","time":"2024-04-04T16:36:38Z","urlPath":"/","uuid":"fdf28b05-4b08-4a77-86b4-989dd0b8fcdb"}
{"X-Forwarded-For":"","activity":true,"context":"fleet-telemetry","level":"info","method":"GET","msg":"socket_connected","network_interface":"","path":"/","time":"2024-04-04T16:36:38Z","txid":"ba87938c-e739-427e-8908-f3fa99c8d26c","user_agent":"Hermes/1.16.2 (vehicle_device)"}
{"activity":true,"context":"fleet-telemetry","duration_sec":0,"level":"info","msg":"socket_disconnected","time":"2024-04-04T16:36:39Z","total":"0"}
2024/04/04 16:36:39 http: panic serving 192.168.130.103:12842: runtime error: invalid memory address or nil pointer dereference
goroutine 92 [running]:
net/http.(*conn).serve.func1()
	/usr/local/go/src/net/http/server.go:1854 +0xbf
panic({0x1304fe0, 0x1f3bed0})
	/usr/local/go/src/runtime/panic.go:890 +0x263
github.com/beefsack/go-rate.(*RateLimiter).Try(0xc000431ae0)
	/go/pkg/mod/github.com/beefsack/go-rate@v0.0.0-20220214233405-116f4ca011a0/rate.go:55 +0xe8
github.com/teslamotors/fleet-telemetry/server/streaming.(*SocketManager).ProcessTelemetry(0xc00028f200, 0xc00028f200?)
	/go/src/fleet-telemetry/server/streaming/socket.go:198 +0x23a
github.com/teslamotors/fleet-telemetry/server/streaming.(*Server).ServeBinaryWs.func1({0x1738090?, 0xc0000c2460?}, 0xc0000be800)
	/go/src/fleet-telemetry/server/streaming/server.go:102 +0x29a
net/http.HandlerFunc.ServeHTTP(0x0?, {0x1738090?, 0xc0000c2460?}, 0x0?)
	/usr/local/go/src/net/http/server.go:2122 +0x2f
net/http.(*ServeMux).ServeHTTP(0x1314880?, {0x1738090, 0xc0000c2460}, 0xc0000be800)
	/usr/local/go/src/net/http/server.go:2500 +0x149
github.com/teslamotors/fleet-telemetry/server/streaming.serveHTTPWithLogs.func1({0x1738090, 0xc0000c2460}, 0xc0000be800)
	/go/src/fleet-telemetry/server/streaming/server.go:73 +0x391
net/http.HandlerFunc.ServeHTTP(0x0?, {0x1738090?, 0xc0000c2460?}, 0x50794e?)
	/usr/local/go/src/net/http/server.go:2122 +0x2f
net/http.serverHandler.ServeHTTP({0xc0002fe630?}, {0x1738090, 0xc0000c2460}, 0xc0000be800)
	/usr/local/go/src/net/http/server.go:2936 +0x316
net/http.(*conn).serve(0xc0003ae870, {0x1738bd8, 0xc00055e780})
	/usr/local/go/src/net/http/server.go:1995 +0x612
created by net/http.(*Server).Serve
	/usr/local/go/src/net/http/server.go:3089 +0x5ed
{"context":"fleet-telemetry","level":"debug","msg":"return_stop_chan","time":"2024-04-04T16:36:39Z"}
{"context":"fleet-telemetry","level":"debug","msg":"writer_done","time":"2024-04-04T16:36:39Z"}
```

if I check the `/api/1/partner_accounts/fleet_telemetry_errors` endpoint, I see this error: 

```json
            {
                "created_at": "2024-04-04T16:47:45.308654633Z",
                "error": "\"webconnection error: read tcp 192.168.0.252:57868->99.80.125.251:443: i/o timeout\" cm_type=stream",
                "error_name": "cloud_manager_error",
                "hostname": "tesla-telemetry.mydomain.com",
                "name": "7dc3946c846c-42bc-a2fc-ec2c1117a1f7",
                "port": "443",
                "txID": "add6841e-f9c2-4ed1-8f89-eb90c8a0b325",
                "vin": "....."
}
```

Have anyone else facing this issue? It could be a miss configuration?

-----

for the reference the servers's config is: 
```json
    {
      "host": "0.0.0.0",
      "port": 443,
      "log_level": "trace",
      "json_log_enable": true,
      "namespace": "tesla.original-vehicle-data.v1",
      "reliable_ack": true,
      "monitoring": {
        "prometheus_metrics_port": 9090,
        "profiler_port": 4269,
        "profiling_path": "/tmp/trace.out"
      },
      "rate_limit": {
        "enabled": false
      },
      "records": {
        "alerts": [
            "logger"
        ],
        "errors": [
            "logger"
        ],
        "V": [
            "logger"
        ]
      },
      "tls": {
        "server_key": "/etc/ssl-tesla/key.key",
        "server_cert": "/etc/ssl-tesla/key.crt"
      },
      "ca": "-----BEGIN CERTIFICATE-----\n...<content of full certificate chain file> ==\n-----END CERTIFICATE-----\n"
    }
```

agbpatro, 2024-04-04:
Have a [PR](https://github.com/teslamotors/fleet-telemetry/pull/142) to fix it but if you want to get around it for now, please set a rate limiter like https://github.com/teslamotors/fleet-telemetry/blob/main/test/integration/config.json#L34
You can tweak around the values which will be applicable to your integration

agbpatro, 2024-04-04:
@slashmili fixing in https://github.com/teslamotors/fleet-telemetry/pull/143 but if you want to handle it in current version, you may use a config like :-

```sh
    {
      "host": "0.0.0.0",
      "port": 443,
      "log_level": "trace",
      "json_log_enable": true,
      "namespace": "tesla.original-vehicle-data.v1",
      "reliable_ack": true,
      "monitoring": {
        "prometheus_metrics_port": 9090,
        "profiler_port": 4269,
        "profiling_path": "/tmp/trace.out"
      },
      "records": {
        "alerts": [
            "logger"
        ],
        "errors": [
            "logger"
        ],
        "V": [
            "logger"
        ]
      },
      "tls": {
        "server_key": "/etc/ssl-tesla/key.key",
        "server_cert": "/etc/ssl-tesla/key.crt"
      },
      "ca": "-----BEGIN CERTIFICATE-----\n...<content of full certificate chain file> ==\n-----END CERTIFICATE-----\n"
    }

```

slashmili, 2024-04-05:
Thanks @agbpatro!

I've removed the `rate_limit` key in the config and it's working.

closed 2024-04-05 by slashmili

rileymd88, 2024-06-10:
I am getting a similar error as well:

```
2024/06/10 20:16:07 http: panic serving 176.199.247.105:42518: runtime error: invalid memory address or nil pointer dereference
goroutine 34 [running]:
net/http.(*conn).serve.func1()
	/usr/local/go/src/net/http/server.go:1854 +0xbf
panic({0x12f8640, 0x1f21ec0})
	/usr/local/go/src/runtime/panic.go:890 +0x263
github.com/sirupsen/logrus.(*Logger).Logf(0x8?, 0xffffffff?, {0x141c9d9?, 0x1?}, {0xc0002c3150?, 0x9?, 0xc000320028?})
	/go/pkg/mod/github.com/sirupsen/logrus@v1.9.0/logger.go:152 +0x22
github.com/sirupsen/logrus.(*Logger).Errorf(...)
	/go/pkg/mod/github.com/sirupsen/logrus@v1.9.0/logger.go:186
github.com/teslamotors/fleet-telemetry/datastore/kafka.(*Producer).logError(0xc0000929a0?, {0x1722de0?, 0xc00041c400?})
	/go/src/fleet-telemetry/datastore/kafka/kafka.go:113 +0x65
github.com/teslamotors/fleet-telemetry/datastore/kafka.(*Producer).Produce(0xc000038040, 0xc0001ec6c0)
	/go/src/fleet-telemetry/datastore/kafka/kafka.go:78 +0x5e9
github.com/teslamotors/fleet-telemetry/telemetry.(*BinarySerializer).Dispatch(...)
	/go/src/fleet-telemetry/telemetry/serializer.go:93
github.com/teslamotors/fleet-telemetry/telemetry.(*Record).Dispatch(0xc0001ec6c0)
	/go/src/fleet-telemetry/telemetry/record.go:100 +0x13a
github.com/teslamotors/fleet-telemetry/server/streaming.(*SocketManager).processRecord(0xc0000eb760?, 0xc0001ec6c0)
	/go/src/fleet-telemetry/server/streaming/socket.go:263 +0x25
github.com/teslamotors/fleet-telemetry/server/streaming.(*SocketManager).ParseAndProcessRecord(0xc0002f7dd0, 0xc00041c360, {0xc00009c600?, 0x18?, 0xc0002c3770?})
	/go/src/fleet-telemetry/server/streaming/socket.go:254 +0x673
github.com/teslamotors/fleet-telemetry/server/streaming.(*SocketManager).ProcessTelemetry(0xc0002f7dd0, 0xc0002f7dd0?)
	/go/src/fleet-telemetry/server/streaming/socket.go:218 +0x605
github.com/teslamotors/fleet-telemetry/server/streaming.(*Server).ServeBinaryWs.func1({0x1727b20?, 0xc0002e8540?}, 0xc000170e00)
	/go/src/fleet-telemetry/server/streaming/server.go:99 +0x2d3
net/http.HandlerFunc.ServeHTTP(0xc00037d5e0?, {0x1727b20?, 0xc0002e8540?}, 0xc0002c3920?)
	/usr/local/go/src/net/http/server.go:2122 +0x2f
net/http.(*ServeMux).ServeHTTP(0xc00009e500?, {0x1727b20, 0xc0002e8540}, 0xc000170e00)
	/usr/local/go/src/net/http/server.go:2500 +0x149
github.com/teslamotors/fleet-telemetry/server/streaming.serveHTTPWithLogs.func1({0x1727b20, 0xc0002e8540}, 0xc000170e00)
	/go/src/fleet-telemetry/server/streaming/server.go:70 +0x236
net/http.HandlerFunc.ServeHTTP(0x0?, {0x1727b20?, 0xc0002e8540?}, 0x50794e?)
	/usr/local/go/src/net/http/server.go:2122 +0x2f
net/http.serverHandler.ServeHTTP({0xc0002adb90?}, {0x1727b20, 0xc0002e8540}, 0xc000170e00)
	/usr/local/go/src/net/http/server.go:2936 +0x316
net/http.(*conn).serve(0xc000279cb0, {0x1728668, 0xc0004d1590})
	/usr/local/go/src/net/http/server.go:1995 +0x612
created by net/http.(*Server).Serve
	/usr/local/go/src/net/http/server.go:3089 +0x5ed
```

My config looks like this:
```
{
  "host": "0.0.0.0",
  "hostname": "mydomain.com",
  "port": 1000,
  "log_level": "debug",
  "json_log_enable": true,
  "namespace": "telemetry",
  "reliable_ack": true,
  "reliable_ack_sources": {
    "V": "kafka"
  },
  "kafka": {
    "bootstrap.servers": "kafka-server:9092",
    "sasl.mechanism": "SCRAM-SHA-256",
    "security.protocol": "SASL_SSL",
    "sasl.username": "username",
    "sasl.password": "password"
  },
  "records": {
    "alerts": ["logger"],
    "errors": ["logger"],
    "V": ["kafka"]
  },
  "tls": {
    "server_cert": "cert.pem",
    "server_key": "private.pem"
  },
  "ca": "-----BEGIN CERTIFICATE-----<content of full cert chain file>-----END CERTIFICATE-----\n"
}
```

The error only happens after adding my kafka config. If I change the `records.V` config entry from `["kafka"]` to `["logger"]` then I do not see this error and I see the messages coming in correctly. 

The results of the `api/1/partner_accounts/fleet_telemetry_errors` call does not shows the following although I believe it is just an error for when my fleet-telemetry container was not running:
```
{
                "created_at": "2024-06-10T20:09:02.175200376Z",
                "error": "\"webconnection error: dial tcp myserverip:1000: connect: connection refused\" cm_type=stream",
                "error_name": "cloud_manager_error",
                "hostname": "mydomain.com",
                "name": "1ed6ad0b3a7d-43e3-9803-33b794e0bd15",
                "port": "1000",
                "txID": "6d388b75-2e23-4c8b-ac57-42aa46e69ab5",
                "vin": "myvin"
            }
```