teslamotors/fleet-telemetry #168: NOTE: GitHub issues temporarily disabled

patrickdemers6, 2024-04-29:
A [security issue](https://www.bleepingcomputer.com/news/security/github-comments-abused-to-push-malware-via-microsoft-repo-urls/#google_vignette) with Github has been used in phishing attempts.

The issue tracker will be re-enabled when Github releases a fix.

In the meantime, please send urgent bug reports to [fleetapisupport@tesla.com](mailto:fleetapisupport@tesla.com). (Please do not use this email address to submit issues for this repository once the issue tracker is re-enabled).

bramp, 2024-05-20:
Looks like issues are re-enabled? or at least I can create a "New Issue", so this bug is now resolved?

patrickdemers6, 2024-05-24:
Yep, GitHub mitigated the issue. Thanks!

closed 2024-05-24 by patrickdemers6