teslamotors/fleet-telemetry #170: some vehicle data questions`

idonkeyliu, 2024-05-17:
1. How can I determine the total battery capacity from the vehicle data?
2. How can I obtain the energy consumption information?
3. How can I distinguish between the new Model 3 and the old Model 3?
4. How can I differentiate between the long-range version and the standard version?

jlestel, 2024-05-30:
Hi @patrickdemers6 

Another question about `vehicle_data`: is the field [`drive_state.power`](https://developer.tesla.com/docs/fleet-api#vehicle_data) available on Telemetry? Can't find it.

patrickdemers6, 2024-05-31:
I don't think it's available at this time. I'll create a note to consider adding it in a future firmware version.

kanhaiya-2000, 2024-06-11:
where can i find details about units of every field ?

idonkeyliu, 2024-06-13:
> 1. You could extrapolate from vehicle_data's `charge_state.battery_range / charge_state.battery_level * 100`
> 2. This is very vague, you can aggregate data however you need. The API provides up-to-date data.
> 3. You should be able to find a field in vehicle data's `vehicle_config` object which indicates. I haven't tested between the two to see the differences.
> 4. Check out the [vehicle options endpoint](https://developer.tesla.com/docs/fleet-api#options)

You could extrapolate from vehicle_data's charge_state.battery_range / charge_state.battery_level * 100
charge_state.battery_range / charge_state.battery_level * 100. 
this can obtain total range for full battery, but can not obtain  battery capacity (kwh)

idonkeyliu, 2024-06-13:
> Hi @patrickdemers6 where can i find details about units of every field in charge_state ? And also details like battery capacity(in kwh) and vehicle's model details ? Can u please provide some help here ?

I have same question too.  Looking forward to receiving a reply.

kanhaiya-2000, 2024-06-30:
I hope i got an approach.

To calculate battery capacity:
We have these fields in charge_state
"charge_energy_added": 47.01,
"charge_miles_added_ideal": 201.0,
"charge_miles_added_rated": 201.0,
These miles and energy added should be proportional to each other.
So ideal battery capacity for a full range =  (charge_energy_added / charge_miles_added_ideal)*(ideal_battery_range / battery_level) * 100

Model details:
vehicle_config.car_type provides this details.

I have put it so that anyone else doesn't go through all the troubles of looking at the vast json carefully.

idonkeyliu, 2024-07-05:
> I hope i got an approach.
>
> To calculate battery capacity: We have these fields in charge_state "charge_energy_added": 47.01, "charge_miles_added_ideal": 201.0, "charge_miles_added_rated": 201.0, [...]

Currently, this is how I am doing it.

closed 2025-01-22 by patrickdemers6