teslamotors/fleet-telemetry #172: Error CSR when fleet_telemetry_config create

themarch, 2024-05-23:
Hi, i have a problem with my CSR when creating the fleet-telemetry config. However, my CSR has been validated because with this third-party token I can use the API in the usual way, like ‘vehicle_data’ for example. 

I've followed all the previous steps to get fleet-telemetry on my server, but I get this error. Thanks in advance

I've noticed that it's only related to my bearer, because I can put anything in the body, it won't change anything. 
Here's my "[URL](bib-data.fr)" : bib-data.fr

[image]

For example, here I have access to the classic API, my CSR has been accepted and I've already received an email about it.

[image]

Thanks in advance.

Martucci98, 2024-05-23:
[image]

To add some information for the request: "{{FLEET_API_URL}}/api/1/vehicles/{{vehicle_tag}}/fleet_telemetry_config"  got a synced but to "true" (see the screen) 

Thank you for your help

patrickdemers6, 2024-05-24:
@themarch you first setup your account in NA region, but then set up in EU. I've applied the signed certificate to your account in EU so you should now be able to setup fleet-telemetry.

patrickdemers6, 2024-05-24:
@Martucci98 you are passing a third-party token to the fleet_telemetry_errors endpoint. It requires a partner token. Please see the [Authentication section](https://developer.tesla.com/docs/fleet-api#authentication) of the docs to learn the differences between these tokens.

themarch, 2024-05-27:
Thanks a lot @patrickdemers6, it's done for this part.

Just one very strange thing. I got this when I wanted to add the public key to the vehicle. However, I do have access to this vehicle, and I can make API calls on it etc. And I've already done the ‘third_party’ and ‘authorization_code’ parts. 

Maybe it's another zone problem? 

[image]

themarch, 2024-05-27:
Thanks a lot @patrickdemers6, it's done for this part.

Just one very strange thing. I got this when I wanted to add the public key to the vehicle. However, I do have access to this vehicle, and I can make API calls on it etc. And I've already done the ‘third_party’ and ‘authorization_code’ parts.

Maybe it's another zone problem? (for the translation of the pictures it's : The vehicle cannot be shared.
Adding a virtual key for ‘bib-data.fr’ is prohibited because you have not granted ‘bib-data.fr’ access to your account.)

[image]

closed 2024-05-29 by themarch