teslamotors/fleet-telemetry #173: Can't add public key to vehicule

themarch, 2024-05-29:
I got this when I wanted to add the public key to the vehicle. However, I do have access to this vehicle, and I can make API calls on it etc. And I've already done the ‘third_party’ and ‘authorization_code’ parts. (also with vehicle_device_data scope)

Maybe it's another zone problem? (for the translation of the pictures it's : The vehicle cannot be shared.
Adding a virtual key for ‘bib-data.fr’ is prohibited because you have not granted ‘bib-data.fr’ access to your account.)

[image]

But as you can see, on the client account, all is link with "bib-data.fr"

[image]

themarch, 2024-05-29:
Okay @patrickdemers6 
I tried with another vehicle (which has a more recent software version and it works), this must have an impact.

patrickdemers6, 2024-05-29:
This error occurs when you haven't gone through all the authorization code flow steps. Vehicle firmware has nothing to do with it.

When you did a different vehicle, was it on a different account?

patrickdemers6, 2024-05-29:
I believe it's also possible this happens when your server can't be accessed from the mobile app when it validates public key.

patrickdemers6, 2024-06-01:
Can you email fleetapisupport@tesla.com with the two email addresses you tried? Make sure to call out which one is working and which one isn't.

SirArkimedes, 2024-06-01:
@patrickdemers6 We've had this occur on almost all of our accounts that both:
1. owned vehicles and
2. were some of the first accounts that authorized our application.

Based on what we were seeing, we kinda suspected that this started occurring around the time that we were able to retroactively change scopes without revoking the application and re-adding it.

Our (hopefully temporary) workaround has been to revoke the application and re-authenticate.

patrickdemers6, 2024-06-01:
Thanks for letting me know. Narrowing down the timeline of this starting helps a lot. If you can also email any additional details to me, it will help with investigating.

themarch, 2024-06-01:
I've asked for the email address of the second account, but I don't have it directly. I'll send you an email as soon as I have all the information, detailing everything. 

We've already tried deleting revoke the application and re-authenticate but no change. 

Thanks in advance!

themarch, 2024-06-01:
I've just emailed you the information! @patrickdemers6

paede81, 2024-06-03:
Faceing the exact same issue, any advice what should I test?
@patrickdemers6 I followed exactly the instruction you provided

patrickdemers6, 2024-06-13:
Sorry for the delay everyone, I'm struggling to identify the issue. Can someone reproduce it again and share VIN/email? I know you already have, but I want to dig into the logs while they're fresh.

cc @themarch @paede81

themarch, 2024-06-17:
Hi @patrickdemers6 
I've just tried it out and same error ! 
For the VIN: LRWYGCFS6PC552861
And email: cv.lapeyre@orange.fr

paede81, 2024-06-17:
Hi @patrickdemers6 
VIN: LRW3E7EK5MC400705
Email: patrick.steiner@gmx.ch 

I have an idea: could the issue be related to the endpoints and regional requirements?

I registered it via
https://fleet-api.prd.na.vn.cloud.tesla.com/
But my car is located in Europe.
After I realised my mistake I raised a new request with
https://fleet-api.prd.eu.vn.cloud.tesla.com/
And wait for the answer from Tesla.

@themarch could it be the same with you, as it seems you live also in Europe?

paede81, 2024-06-17:
> Sorry for the delay everyone, I'm struggling to identify the issue. Can someone reproduce it again and share VIN/email? I know you already have, but I want to dig into the logs while they're fresh.
> 
> cc @themarch @paede81

Tried it right now but still the same issue

patrickdemers6, 2024-06-17:
Thanks all.

@paede81 I don't see that you've granted an application access to that email. Can you confirm you've gone through full authorization flow?

@themarch you have granted that email address access to the application; however, it looks like the mobile app is signed in with an email beginning with `pa`. Can you make sure you're using the same email for authorization flow and in the mobile app?

themarch, 2024-06-20:
I think that's it @patrickdemers6 
I'll give it a go, but in fact: the tesla is X with X account. 

But I sent the QRCode to be scanned to a secondary driver with email Y. I'm testing X with X later today.

themarch, 2024-06-26:
@patrickdemers6 Sorry to delay, it was that ! Thanks a lot

paede81, 2024-08-23:
> Thanks all.
>
> @paede81 I don't see that you've granted an application access to that email. Can you confirm you've gone through full authorization flow?
>
> @themarch you have granted that email address [...]

Hi Patrick (@patrickdemers6)
Sorry for the late response. Yes I did go through the full authorization flow but I probably used the wrong API hostname as I live in Europe and I used the US one. In the meantime I started again with the correct URL and seems it's working now to add the public key to the vehicle. But now facing the next issue: Send Fleet Telemetry Config not accepted as CSR not recognized by Tesla. Could this issues also be related to the other issue? Send first the CSR to the US host and then again to the EU host?
`
    "response": null,
    "error": "Please follow instructions https://developer.tesla.com/docs/fleet-api#fleet-telemetry to send a CSR to Tesla; configuration may take up to two weeks.",
    "error_description": "",
    "txid": "579701de39f540145145d4ca683e960e"
`

patrickdemers6, 2024-08-24:
@paede81 I updated your EU account so it is configured for streaming.

Closing this issue.

closed 2024-08-24 by patrickdemers6