teslamotors/fleet-telemetry #174: check_server_cert.sh not validate server certificate

krishR007, 2024-05-29:
error 20 at 0 depth lookup: unable to get local issuer certificate
error /tmp/tmp.wJir4samcA: verification failed
CN = elvee.app
error 20 at 0 depth lookup: unable to get local issuer certificate
error /tmp/tmp.wJir4samcA: verification failed
-e The server certificate is invalid.

patrickdemers6, 2024-05-29:
Please describe your setup, steps taken, etc.

krishR007, 2024-05-30:
1. After tesla fleet confirmation I generate SSL certificate on server:
https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-nginx-in-ubuntu

2. Validate_server.json not validating using check_server_cert.sh. My config file is below:

{
  "hostname": "MY_HOST_NAME",
  "port": "443",
  "ca": "MY_CERTIFICATE_CONTENT"
}

krishR007, 2024-06-03:
Can you please help me to solve this issue ?

closed 2024-06-05 by krishR007