teslamotors/fleet-telemetry #179: Fleet Telemetry and Kafka

Urkman, 2024-06-16:
I think, I got every part working.
Now I'm trying to stick everything together.

I installed Kafka on a Digital Ocean Droplet and this is working.
I can send messages from the producer to the consumer. Both using localhost:9092.

Fleet Telemetry is setup on the same machine using docker.
This is also working, as I get events from my Tesla.

I used this article for the Fleet Telemetry installation:
https://dev.to/patrickdemers6/fleet-telemetry-streaming-data-from-your-tesla-2i97

But, how can I connect Kafka and the Fleet telemetry?

I tried wit two configs, both ending in errors.

First config:

```
  "kafka": {
    "bootstrap.servers": "localhost:9092",
    "queue.buffering.max.messages": 10000
  },
```
When I start the Telemetry Server I get this error:

```
[thrd:localhost:9092/bootstrap]: localhost:9092/bootstrap: Connect to ipv4#127.0.0.1:9092 failed: Connection refused (after 4ms in state CONNECT)
```

And with this config::

```
  "kafka": {
    "bootstrap.servers": "kafka:9092",
    "queue.buffering.max.messages": 10000
  },
```

When I start the Telemetry Server I get this error:

```
[thrd:kafka:9092/bootstrap]: kafka:9092/bootstrap: Failed to resolve 'kafka:9092': Temporary failure in name resolution (after 7ms in state CONNECT)
```
I start the docker container with `docker compose up`

What I'm doing wrong here?
Any tipps?

rileymd88, 2024-06-17:
Try this:
<your_droplet_ip>:9092

krishR007, 2024-08-13:
After changes , Have you received data on your own server ?

closed 2024-08-24 by patrickdemers6