teslamotors/fleet-telemetry #180: command fleet_status / Http Status Code 421 (MisdirectedRequest)

bassmaster187, 2024-06-16:
I receive sometime Http Statuscode 421 on command fleet_status. What does that mean?

Bre77, 2024-06-16:
`This user is not present in the current region. See the regional requirements section for more information.`

https://developer.tesla.com/docs/fleet-api#response-codes

bassmaster187, 2024-06-16:
Maybe related to this issue:
https://github.com/teslamotors/vehicle-command/issues/249

Bre77, 2024-06-16:
> Maybe related to this issue:
> https://github.com/teslamotors/vehicle-command/issues/249
> 

To mitigate that I call `api/1/users/region` for every user login to ensure the region is correct. A month ago I had a handful of EU users showing up as NA, although that hasn't happened for a while.

bassmaster187, 2024-06-17:
The Proxy Server should handle all the region stuff. At least for all other users.

patrickdemers6, 2025-01-22:
I don't believe fleet_status raises 412 anymore. If you're still having issues, let me know. Also feel free to create an issue in vehicle-command if region-related logic would still be useful.

closed 2025-01-22 by patrickdemers6