teslamotors/fleet-telemetry #188: Add event for connect and disconnect

Bre77, 2024-07-07:
It would be a huge value to have the Fleet Telemetry server emit an event when a vehicle connects and disconnects, so that a third party service knows if a vehicle is awake or asleep/offline, and avoid polling the Fleet API when a vehicle is asleep or to find out when it wakes up.

I can see these events written to standard out, but they are not emitted through the event buses.

rileymd88, 2024-07-07:
This is already supported. If the vehicle is awake and you are subscribed
to some fields then it will send data. When the car stops sending data then
it is asleep

Bre77, 2024-07-07:
> This is already supported. If the vehicle is awake and you are subscribed
> to some fields then it will send data. When the car stops sending data then
> it is asleep

That is incorrect, certain telemetry fields are only sent under certain conditions, so when you might think a vehicle is "asleep" might actually be the vehicle in park.

patrickdemers6, 2025-01-22:
Closing as this has been implemented.

Related though, there are reliability issues: #244

closed 2025-01-22 by patrickdemers6