teslamotors/fleet-telemetry #189: Higher Limits for Open-Source Applications

SebastianGode, 2024-07-09:
With the introduction of the new Open-Source category for projects I'm a bit shocked that those projects still have the same limits as the discovery plan and that there's no way to even upgrade them.
https://developer.tesla.com/docs/fleet-api#2024-07-08-open-source-support

Overall I think Tesla benefits from the Open-Source community and many owners just want e.g. some driving stats of their vehicle. They don't want to make money from it.

I would really like if the OpenSource tier can get higher API limits than the DiscoveryPlan. I understand, that the API usage also isn't free to run and that Tesla of course needs to charge big companies for using it, but in my opinion the amount of Open-Source apps like Teslamate really should still be usable with the new Streaming-API. If you don't offer options to pay for the API, please at least increase the limits a bit so that it still is possible to run such loggers.

TLDR: I really would like to have a separate plan for Open-Source applications with higher limits than the default DiscoveryTier for companies.

Bre77, 2024-07-10:
Firstly, thank you for raising this issue and making me aware Tesla had added this new open source category.

I have implemented the OAuth flows in Home Assistant and the rate limits are making it rather useless indeed. For some reason even polling every 5 minutes was still being rate limited, and I think I can't even have the whole interaction start up at once because it takes 4 API calls for 1 vehicle and 1 energy site, which also gets me blocked. (Products, vehicle_data, site_info, live_status)

I also don't know how command signing is meant to work but I'll raise that issue separately.

netdata-be, 2024-07-10:
I also believe that this will be the end for projects like teslamate :-(
The rate limit on the API is actually OK if the limit of the Telemetry would be higher.
Currently the undocumented web socket API is nearly realtime, the current discovery tier is 10 items / min which is really limited and not enough to achieve the same as we currently have with the owners api in conjunction with the websocket streaming.

Any suggestions from Tesla side on this?

juergen852, 2024-08-08:
**Main reason for me buying a Tesla** is free Software like Teslalogger .

If API remains restricted like this, there are many reasons for buying at VW, Audi, BMW....
Like much better lane assist, auto parking into garage, autopilot at more than 140 km/h,  no phantom breakings, better ventilation distribution, more luxury feeling inside, and so on....

netdata-be, 2024-11-29:
@SebastianGode I think you can close this issue, because since the new pricing this is kind of "solved"

closed 2024-11-29 by SebastianGode