teslamotors/fleet-telemetry #192: Changes on 2024.26.3.x vehicles?

Urkman, 2024-07-30:
I'm experiencing data changes on 2024.26.3.x vehicles like this:

Location comes as: "(49.851094 N, 6.894920 E)"

And everything else is formatted like this:
`„\"81.906\"“` und nicht mehr `„81.906“`

For vehicles with 2024.20.9 everything is fine...

Urkman, 2024-07-30:
Here is a log entry for a VIN I know it's on 2024.26.3.1

> time="2024-07-30T15:26:52Z" level=info msg="logger_json_unmarshal VIN map[receivedat:1722353212000 timestamp:0 txid: txtype:V version:0 vin: VIN] {\"data\":[{\"key\":\"Location\", \"value\":{\"stringValue\":\"\\\"(49.850634 N, 6.894050 E)\\\"\"}}, {\"key\":\"FastChargerPresent\", \"value\":{\"stringValue\":\"\\\"false\\\"\"}}, {\"key\":\"ChargeState\", \"value\":{\"stringValue\":\"\\\"Idle\\\"\"}}, {\"key\":\"Gear\", \"value\":{\"stringValue\":\"\\\"R\\\"\"}}], \"createdAt\":\"2024-07-30T15:26:52.017841265Z\", \"vin\":\"VIN\"}\n"

bassmaster187, 2024-07-30:
It will be very hard to maintain a bunch ob parsers, logic etc for different firmware versions if tesla will make brakeing changes from time to time...

krishramani01, 2024-08-06:
Can you tell how to decode data from we socket?

Bre77, 2024-08-07:
I also have a user with this exact firmware and getting this different data structure. @patrickdemers6 is this a permanent change or a bug in the vehicle firmware?

Urkman, 2024-08-09:
Now that we have 2024.26.6 there are some changes back to older data format.
But there is one huge problem.

Sending a new Config is not working most of the time. 
When I send a new config it is successful and checking if the config on the car is fine is also successful.
But it keeps sending the old data for ever :( 
@patrickdemers6 can you please take a look at this. This will stop my App completely from woking...

patrickdemers6, 2024-08-24:
The change in data format was unintentional and has been resolved beginning with firmware version `2024.26.4`.

@Urkman let me follow up in the other issue you keep pinging me in.

closed 2024-08-24 by patrickdemers6