teslamotors/fleet-telemetry #194: "unknown user" for Tesla4Business account.

themarch, 2024-08-02:
Hello, 

We have a process that works by sending the link, then the user validates the accesses, we get the code then we have a bearer and a refresh. 

Except that for people with a "**Tesla4Business**" account, we have a code then bearer / refresh, except that at each API call with this bearer we get: "unknown user". Is this because it's a business account? Do we have to use fleet-telemetry rather than the API? Thanks in advance @patrickdemers6 !

[image]

themarch, 2024-09-06:
> This happens when the account has never used the mobile app before. This
> is most common for T4B accounts. Can you try having the same user login
> there?
>
> I know this isn't ideal - working on a long term solution.

Hi, we asked the person to do this process but it was inconclusive.

We always get the same mistake.
Here is the user code:
EU_5037e8e358993e6dcde4bda62494174d142ae6e3785ccb04e584fcf25b4d

The bearer:
eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCIsImtpZCI6InE0dHg3Q1UyYzI2V1BiemwxZjZjanM3QnhzayJ9.
eyJpc3MiOiJodHRwczovL2F1dGgudGVzbGEuY29tL29hdXRoMi92My9udHMiLCJhenAiOiI4ODMyMjc3YWU0Y2MtNDQ2MS04Mzk2LTEyNzMxMDEyOWRjNiIsInN1YiI6Ijc0NGY3MWEwLTI0YzAtNDMxYy05ZGMyLWE0MDI1NTgwYTk4NyIsImF1ZCI6WyJodHRwczovL2ZsZWV0LWFwaS5wcmQubmEudm4uY2xvdWQudGVzbGEuY29tIiwiaHR0cHM6Ly9mbGVldC1hcGkucHJkLmV1LnZuLmNsb3VkLnRlc2xhLmNvbSIsImh0dHBzOi8vYXV0aC50ZXNsYS5jb20vb2F1dGgyL3YzL3VzZXJpbmZvIl0sInNjcCI6WyJvcGVuaWQiLCJvZmZsaW5lX2FjY2VzcyIsImVuZXJneV9kZXZpY2VfZGF0YSJdLCJhbXIiOlsicHdkIl0sImV4cCI6MTcyNTY1NTg3NiwiaWF0IjoxNzI1NjI3MDc2LCJvdV9jb2RlIjoiRVUiLCJsb2NhbGUiOiJkZS1ERSIsImFjY291bnRfdHlwZSI6ImJ1c2luZXNzIiwib3Blbl9zb3VyY2UiOm51bGwsImFjY291bnRfaWQiOiI4ZDc0ZjFlMS05NDM3LTQwZmItYjYwYi04ODFjZGVlMmQ2MmYiLCJhdXRoX3RpbWUiOjE3MjU2MjcwNzV9.
dv1gMe5NH43tk8iJQgMZrV3UxoMJyGNcibNP2nVpHeKgbzDNvplHZcLPnNxLbBXzL-Vs7qnw8khuGDlCoyGBbDfFAoDo0na2tFmiluS4fAtfmFNmNOh_6zw8CABg02F3vTSHPawCMC7Uu7vOEEehbrhwO8cvUulqScLwaF8HRvm6YeLY6i2P-
zViJ8xjOwru2KmfJVb8oJnakuXYaOCQ74H-c4ufbdkUoojkG8PNDLgp1AZlLTBQb61fPC12gL__KxkUuCPwecVUEbVKkiFIYphmMktkE6L0-TyHdnm4tWPbXcbeIqpQRLjJ4vTXgHI_Q3JB164P6b6tSnoNjz2NIQ

And the refresh:
EU_dd48fb5f8cf1cd4a77a3061bc0c6988f5804e77c43ad9be50220f68c9c030d13

rileymd88, 2024-09-06:
You should remove the users token here ASAP

themarch, 2024-09-06:
damn i thought i was by email. thanks for the alert

closed 2025-01-22 by patrickdemers6