teslamotors/fleet-telemetry #195: Decode streaming data

krishramani01, 2024-08-06:
Hello, I've created web socket server in nodejs and telemetry connected successfully and data getting in buffer format and I'm trying to decode this data to original data but getting unreadable format.please check screen shot 
[image]

Bre77, 2024-08-06:
The data is sent as protobuf, so you need to decide the raw data using the protobuf definitions in this repo.

krishramani01, 2024-08-07:
Thanks everything work fine data getting in telemetry server but my question is how to get this data in my server?

Bre77, 2024-08-07:
You need to learn how to handle protobufs, and implement it in your server using the proto files from this repo. Here is a result I got from a google search. https://thecodebarbarian.com/working-with-protobufs-in-node-js.html

You could also try asking your question on StackOverflow or to an LLM chatbot.

krishramani01, 2024-08-07:
Is it possible to get telemetry data in my server using Kafka?

closed 2024-08-17 by krishramani01