teslamotors/fleet-telemetry #197: Kafka REST Proxy error Unexpected exception in consumer read: Unexpected character ('Y' (code 89)): Expected space separating root-level values

rmgodbole, 2024-08-12:
Environment: docker with cp-kafka:7.3.6 with Kafka REST Proxy cp-kafka-rest:7.2.6

Vehicle successfully registered and sending data to tesla fleet telemetry server.

1) Create consumer group and instance
curl -X POST -H "Content-Type: application/vnd.kafka.v2+json" --data '{"name": "my_consumer_instance", "format": "json", "auto.offset.reset": "earliest"}' http://localhost:8082/consumers/my_consumer_group
2) Subscribe to consumer instance with the topic tesla_telemetry_V
curl -X POST -H "Content-Type: application/vnd.kafka.v2+json" --data '{"topics":["tesla_telemetry_V"]}' http://localhost:8082/consumers/my_consumer_group/instances/my_consumer_instance/subscription
3) Fetch records from consumer instance
curl -X GET -H "Accept: application/vnd.kafka.json.v2+json" http://localhost:8082/consumers/my_consumer_group/instances/my_consumer_instance/records

{"error_code":40801,"message":"com.fasterxml.jackson.core.JsonParseException: Unexpected character ('Y' (code 89)): Expected space separating root-level values\n at [Source: (byte[])\"<<<<VIN that starts with 5Y >>>>>>\"; line: 1, column: 3]"}%

krishR007, 2024-08-14:
Can you tell me how to get data in our own server using kafka ?

rmgodbole, 2024-08-20:
> Can you tell me how to get data in our own server using kafka ?

1) Deploy cp-kafka-rest and schema-registry in addition to kafka using docker.  

These docker-compose file examples might help https://github.com/teslamotors/fleet-telemetry/blob/main/docker-compose.yml and https://github.com/confluentinc/cp-all-in-one

2) Make REST requests to your kafka-rest server using binary format
https://docs.confluent.io/platform/current/kafka-rest/quickstart.html?&_ga=2.125942536.441766527.1723856151-1458415198.1723089502&_gl=1*ycmkv3*_gcl_au*ODMwNzE3MzEzLjE3MjMwODk1MDI.*_ga*MTQ1ODQxNTE5OC4xNzIzMDg5NTAy*_ga_D2D3EGKSGD*MTcyMzg1NjE1MS4zLjEuMTcyMzg1NjE4OC4yMy4wLjA.#produce-and-consume-binary-messages

3) Decode the returned "key" and "value" fields using protobuf

patrickdemers6, 2024-08-21:
@rmgodbole can you share the configuration file for your `fleet-telemetry` instance? I want to make sure you're producing json, not protos (which is the default).

rmgodbole, 2024-08-21:
> @rmgodbole can you share the configuration file for your `fleet-telemetry` instance? I want to make sure you're producing json, not protos (which is the default).

[config_redacted.json](https://github.com/user-attachments/files/16685143/config_redacted.json)

patrickdemers6, 2024-08-21:
@rmgodbole looks like you're transmitting the proto representation.

If you set `"transmit_decoded_records": true`, the data should be sent as a JSON string.

The `json_log_enable` option is used when formatting logs to stdout.

rmgodbole, 2024-08-21:
@patrickdemers6 Thanks for the info.

patrickdemers6, 2024-08-24:
Closing this as hopefully your issue is resolved. If you're still facing issues with how data is sent, we can reopen.

closed 2024-08-24 by patrickdemers6