teslamotors/fleet-telemetry #199: Requested Fields being received in chunks vs a single batch

andrewdevelopz, 2024-08-23:
I am creating a config using `/api/1/vehicles/fleet_telemetry_config` endpoint and requesting 147 [Fields](https://github.com/teslamotors/fleet-telemetry/blob/main/protos/vehicle_data.proto), all with `interval_seconds` set to 60.

I am expecting a message to include all the requested Fields, once per minute.
 
There is actually one message in the stored data that have all Fields requested. However, when inspecting the data every other message has Fields coming in chunks. So a message is received every few seconds with 1-10 Fields per message.

Would someone please be able to help so that I can receive all the requested Fields in a single message per minute?

Example Logs:
```
...
{
    "createdAt": "2024-08-23 16:46:27",
    "0": {
        "key": "ScheduledChargingStartTime",
        "value": "some-value"
    }
}
{
    "createdAt": "2024-08-23 16:46:28",
    "0": {
        "key": "ChargePortLatch",
        "value": "some-value"
    }
}
{
    "createdAt": "2024-08-23 16:46:29",
    "0": {
        "key": "ChargeState",
        "value": "some-value"
    },
    "1": {
        "key": "ChargerPhases",
        "value": "some-value"
    }
}
...
```

Bre77, 2024-08-23:
Are you on a paid subscription? Because I thought Fleet Telemetry was limited to 20 fields with 10 values per minute, and you're exceeding that significantly.

andrewdevelopz, 2024-08-23:
My mistake, i am using the max Fields allowed.

patrickdemers6, 2024-08-24:
Data values are sent when the value has changed and the specified interval has passed.

andrewdevelopz, 2024-08-24:
@patrickdemers6 Ah I see, understood. Thank you for your response.

closed 2024-08-24 by patrickdemers6