teslamotors/fleet-telemetry #205: Telemetry stopped sending all requests fields

thesammakman, 2024-08-28:
Starting August 25th we started seeing that instead of getting the entire telemetry payload from vehicles we were receiving only a single field we requested. We originally onboarded with a CSR but I have since switched to using the command proxy to submit telemetry configs, this made no difference in the data we receive. I have also noticed that we are not getting all the fields we requests even as individual messages from the vehicle.

My questions are 
- Is this the expected behavior of the fleet telemetry service? 
- Is there any way we can get the vehicle to send all fields instead of a single one? 
- Should we expect each field to individually update when its value changes on the vehicle? 

Example of what we are receiving right now
``` 
data {
  key: ACChargingEnergyIn
  value {
    string_value: "20.027"
  }
}
created_at {
  seconds: 1724612094
  nanos: 16950365
}
vin: "XXXXXXXXXXX"
```
Example of the payload we were receiving before the 25th. 
```
data {
  key: DCChargingPower
  value {
    string_value: "0.099"
  }
}
data {
  key: ACChargingEnergyIn
  value {
    string_value: "33.540"
  }
}
data {
  key: BatteryLevel
  value {
    string_value: "40.982"
  }
}
data {
  key: ChargeLimitSoc
  value {
    string_value: "100"
  }
}
data {
  key: DCChargingEnergyIn
  value {
    string_value: "33.540"
  }
}
data {
  key: Version
  value {
    string_value: "2024.26.10"
  }
}
data {
  key: VehicleName
  value {
    string_value: "Tron"
  }
}
data {
  key: Odometer
  value {
    string_value: "5495.952"
  }
}
data {
  key: ChargeAmps
  value {
    string_value: "48"
  }
}
data {
  key: ServiceMode
  value {
    string_value: "false"
  }
}
data {
  key: ChargeState
  value {
    string_value: "Idle"
  }
}
data {
  key: ACChargingPower
  value {
    string_value: "0.000"
  }
}
data {
  key: Location
  value {
    location_value {
      latitude: 00.00000
      longitude: 00.00000
    }
  }
}
data {
  key: VehicleSpeed
  value {
    string_value: "69.000"
  }
}
data {
  key: BMSState
  value {
    string_value: "Drive"
  }
}
created_at {
  seconds: 1724612040
  nanos: 88456359
}
vin: "XXXXXXXXXXXXX"
```

Telemetry Config
```
{
    "vins": [
      "XXXXXXXXXXXXXXX"
    ],
    "config": {
    "hostname":"example.com",
    "ca":"-----BEGIN CERTIFICATE-----\n
XXXXXXXXXXXXXXXXX-----END CERTIFICATE-----",
      "fields": {
        "ACChargingPower": {"interval_seconds": 60},
        "ChargeState": {"interval_seconds": 60},
        "Location": {"interval_seconds": 60},
        "Soc": {"interval_seconds": 60},
        "BatteryLevel": {"interval_seconds": 60},
        "TimeToFullCharge": {"interval_seconds": 60},
        "VehicleSpeed": {"interval_seconds": 60},
        "Version": {"interval_seconds": 60},
        "Odometer": {"interval_seconds": 60}
      },
      "alert_types": [
        "service"
      ],
      "exp": 1738696778
    }
  }
```

rileymd88, 2024-08-28:
It was mentioned in another issue that this new behavior is working as
designed now.

andrewdevelopz, 2024-08-28:
Yes, if you see https://github.com/teslamotors/fleet-telemetry/issues/199, only the Fields that have changed in value will be sent in the message.

thesammakman, 2024-08-28:
Should we expect all updated fields in a single message per vehicle or will we get multiple message per vehicle each interval?

andrewdevelopz, 2024-08-28:
I think multiple message per vehicle each interval. From what I can tell the first one has all the fields and every message after that   has only fields that values were changed.

So like:
2024-08-28 00:00:00 - { Odometer, VehicleSpeed, DCChargingPower,  BatteryLevel, ChargeLimitSoc }
2024-08-28 00:01:00 - { BatteryLevel, VehicleSpeed, Odometer }
2024-08-28 00:02:00 - { BatteryLevel, VehicleSpeed }
2024-08-28 00:02:05 - { ChargeLimitSoc, DCChargingPower }

closed 2024-08-29 by thesammakman