teslamotors/fleet-telemetry #207: Fields with no associated value in message body

thesammakman, 2024-08-29:
I am seeing that some of the fields in the message payload no don't include any related values. I just deployed the latest version of the fleet-telemetry to ensure that our infrastructure is not out of date. I am always seeing this with `VehicleSpeed` and have seen it with a  few other fields as well. Here is an example of the payload I'm receiving. 

```
{
    "createdAt": "2024-08-29T19:41:12.288747421Z",
    "data": [
        {
            "key": "VehicleSpeed"
        }
    ],
    "vin": "XXXXXXXXXXXXXX"
}

```

Here is a different payload where you can see VehicleSpeed, ChargerPhases  and VehicleName are lacking values. 
```
{
    "payload": {
        "createdAt": "2024-08-29T20:33:49.859360172Z",
        "data": [
            {
                "value": {
                    "stringValue": "44964.097"
                },
                "key": "Odometer"
            },
            {
                "key": "VehicleSpeed"
            },
            {
                "value": {
                    "stringValue": "Idle"
                },
                "key": "ChargeState"
            },
            {
                "key": "ChargerPhases"
            },
            {
                "value": {
                    "stringValue": "0.000"
                },
                "key": "ChargeAmps"
            },
            {
                "key": "TimeToFullCharge"
            },
            {
                "value": {
                    "stringValue": "99"
                },
                "key": "ChargeLimitSoc"
            },
            {
                "value": {
                    "stringValue": "0.000"
                },
                "key": "DCChargingPower"
            },
            {
                "key": "VehicleName"
            },
            {
                "value": {
                    "stringValue": "57.521"
                },
                "key": "BatteryLevel"
            },
            {
                "value": {
                    "stringValue": " "
                },
                "key": "Version"
            },
        ],
        "vin": "XXXXXXXXXXXXX"
    }

```

patrickdemers6, 2024-08-30:
I know you said latest, but can you confirm the version of fleet-telemetry you're running?

thesammakman, 2024-08-30:
I'm pulling the `latest` [tag from dockerhub.](https://hub.docker.com/r/tesla/fleet-telemetry/tags#!) It appears to be pointed at version v0.3.1 right now. I just pulled it again today, redeployed and am still seeing the error. I can try to use the tag`v0.3.1` to see if that makes a difference.

thesammakman, 2024-08-30:
I just explicitly deployed `v0.3.1` and I'm still seeing the issue 
```
{
key
	
VehicleSpeed
},
```

patrickdemers6, 2024-08-30:
Can you share a bit more about your setup? A redacted config.json file would be great.

thesammakman, 2024-08-30:
We are pulling the container and deploying to AWS ECR Fargate. We using the vended image as a base image and building on it to add the config.json and tls cert. We are connecting the service to an AWS hosted kafka instance. 

```
{
    "host": "0.0.0.0",
    "port": 443,
    "log_level": "info",
    "json_log_enable": true,
    "namespace": "tesla_fleet_telemetry",
    "reliable_ack": false,
    "status_port":80,
    "kafka": {
        "bootstrap.servers": "*******",
        "queue.buffering.max.messages": 1000000,
        "security.protocol": "SSL"
    },
    "rate_limit": {
        "enabled": true,
        "message_interval_time": 30,
        "message_limit": 1000
    },
    "records": {
        "alerts": [
            "logger"
        ],
        "errors": [
            "logger"
        ],
        "V": [
            "kafka"
        ]
    },
    "tls": {
        "server_cert": "/etc/certs/server/tls.crt",
        "server_key": "/etc/certs/server/tls.key"
    }
  }
```

patrickdemers6, 2024-08-30:
You're sending proto messages to Kafka. In the consumer, have you updated the proto to the latest version? We recently introduced `invalid` type so these values may be returning Invalid. For a field like `VehicleSpeed`, it should turn valid once the vehicle is being driven.

thesammakman, 2024-08-30:
That's probably the issue. Our proto was out of date. I'll have to monitor to confirm that we are not seeing the issue anymore. Thanks for the help!

thesammakman, 2024-09-03:
Updating the proto fixed this issue. Thanks for the support!

closed 2024-09-03 by thesammakman