teslamotors/fleet-telemetry #210: Events are becoming more and more fragmented since 08/10

jlestel, 2024-09-09:
As you can see below, the events become more and more fragmented since around August 10th.
Previously, most of the requested fields were found in each event and size was 700b in average.

Is there a reason for this change? How can we switch to the previous behavior?

[image]

corsair, 2024-09-13:
My understanding is that after a recent software update (it is unclear if 2024.26.X or another build), fields will only be streamed if a change occurs. This reduces overall bandwidth usage, but it may require modifications to data processing to better accommodate this.

jlestel, 2024-10-02:
Indeed, only differential data is sent now. I adapted my script :)

Issue can be closed, ty @corsair

corsair, 2024-10-22:
@jlestel You should be able to close it on your own manually! 👍

jlestel, 2024-10-24:
No i can't :(