teslamotors/fleet-telemetry #215: Does anyone know where FSD version value is coming from on 3rd party sites?

mlancer, 2024-09-26:
This probably isn’t the place to ask but I’m been scratching my head with this for a while. I see 3rd party sites labeling updates with like “FSD 12.3.6”. Maybe I'm overlooking it entirely but can’t find any endpoint in the fleet api or any field from telemetry that actually provides this value. Can’t anyone point me in the direction on where this is located?

corsair, 2024-10-20:
This information is unavailable via Fleet Telemetry or as a directly accessible value.

It can be obtained by calling the [release notes Fleet API endpoint](https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-endpoints#release-notes), locating any release notes mentioning Full Self-Driving, and then extracting the FSD version from there.

Otherwise, I'm sure most services manually do this.

mlancer, 2024-10-21:
> This information is unavailable via Fleet Telemetry or as a directly accessible value.
>
> It can be obtained by calling the [release notes Fleet API [...]

Thank you! I didn't have FSD until this latest demo and I'm seeing what you mean now. Thanks.

closed 2024-10-21 by mlancer