teslamotors/fleet-telemetry #217: Problems Registering - still saying third party not registered 28 days after calling register successfully?

stevedurbinwales, 2024-09-28:
Apologies, I think this should be a discussion, but those are turned off. 

Set up Telemetry with public key visible on server as per guidance, checked the CSR etc. Tried to register, came back successful back in August, and every time I call now it shows August date, but vehicle endpoints are still not active and no signed key back from Tesla. 

I'm getting a partner token, then using that as a bearer to call the register endpoint /api/1/partner_accounts with

{
            "domain": "telemetry.excathedra.solutions",
            "csr": "<csr as a string with \n for newlines"
            }

I'm getting the same response each time which seems to imply my call back in August completed the registration:

{"response":{"account_id":"<uuid>","domain":"telemetry.excathedra.solutions","name":"Internal Vehicle monitoring","description":"Internal data collection and management","issuer":null,"csr":null,"client_id":"<uuid>","ca":null,"created_at":"2024-08-29T22:13:36.400Z","updated_at":"2024-08-30T00:01:24.814Z","enterprise_tier":"discovery","csr_updated_at":null,"public_key":<key>"}}

Can anyone guide me where I'm going wrong here? Have I missed something?

ciumagcatalin, 2024-10-07:
Hi Steve,

Per last tesla announcements the CSR registration is not required anymore (https://developer.tesla.com/docs/fleet-api/support/announcements ). Documentation was updated recently and does not have the CSR point mentioned. I also set my server recently and you don't need to wait for Tesla to get confirmation on CSR.
Continue with the steps and it should work.  
If you are stuck on other step post here as I also struggled a bit to get it work, but I got a bit of help from community .

Catalin

stevedurbinwales, 2024-10-07:
> Per last tesla announcements the CSR registration is not required anymore (https://developer.tesla.com/docs/fleet-api/support/announcements ). 

Thanks Catalin! Very confusing because the documentation still referenced it when I started. I shall try the vehicle command route.

closed 2024-10-07 by stevedurbinwales