teslamotors/fleet-telemetry #218: PubSub does not work

ciumagcatalin, 2024-10-02:
Hello,

I tried to configure the google PubSub dispatcher and it does not seems to run. I tested the connection to Google Cloud separatedly and credentials work fine. The docker logs does not show anything running in terms of this. How can I debug it?
The telemetry receives data fine .

This is my yml config of docker implementation:

_services:
  app:
    build:
      context: ./repo
    ports:
      - 0.0.0.0:443:443
    environment:
      - GOOGLE_APPLICATION_CREDENTIALS=/etc/tesla-telemetry/service-account.json
    volumes:
      - ./certs:/etc/tesla-telemetry/certs:rw
      - ./config.json:/etc/fleet-telemetry/config.json
      - ./service-account.json:/etc/tesla-telemetry/service-account.json:ro_

Config.json:

  _"port": 443,
  "log_level": "debug",
  "json_log_enable": true,
  "namespace": "telemetry",
  "reliable_ack": false,
  "rate_limit": {
    "enabled": false,
    "message_limit": 100
  },
  "records": {
    "alerts": [
        "logger"
    ],
    "errors": [
        "logger"
    ],
    "V": [
          "logger"
      ]
  },
  "pubsub": {
       "gcp_project_id": "tesla-telemetry-437xxxx"_

agbpatro, 2024-10-02:
Hi there
Have you tried setting up the way described in https://github.com/teslamotors/fleet-telemetry/pull/62#issue-1878023909?

ciumagcatalin, 2024-10-02:
If I read it correctly that is for deploying the telemetry on GCP, while I have mine on AWS EC2. If somebody can confirm if the mount is right for the google environment: - ./service-account.json:/etc/tesla-telemetry/service-account.json:ro.
Also, I don't see anything in the logs, so I am not sure if the PubSub is invoked at all... it might be a config thing done wrong.

Bre77, 2024-10-02:
@ciumagcatalin your config is wrong, as per agbpatro's comment, you need to actually enable pubsub in your records too.

`"records": {
        "alerts": [
            "pubsub"
        ],
        "errors": [
            "pubsub"
        ],
        "V": [
            "pubsub"
        ]
    },`

ciumagcatalin, 2024-10-03:
> @ciumagcatalin your config is wrong, you need to actually enable pubsub in your records too.
> 
> `"records": { "alerts": [ "pubsub" ], "errors": [ "pubsub" ], "V": [ "pubsub" ] },`
> 
> This would be why you dont have any errors.

Thank you @Bre77 ! That was missing. Now everything seems to be working!

closed 2024-10-04 by agbpatro