teslamotors/fleet-telemetry #219: Kafka issue

dorrod-blip, 2024-10-02:
Hello,
I'm trying to install fleet-telemetry server on linux and that is the logs.

app-1         | 2024/10/02 16:03:16 maxprocs: Leaving GOMAXPROCS=2: CPU quota undefined
app-1         | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"starting_server","time":"2024-10-02T16:03:16Z"}
app-1         | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"status_server_configured","time":"2024-10-02T16:03:16Z"}
app-1         | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"profiler_started","port":4269,"time":"2024-10-02T16:03:16Z"}
app-1         | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"kafka_registered","namespace":"tesla_telemetry","time":"2024-10-02T16:03:16Z"}
app-1         | {"SocketServer":{"DispatchRules":{"tesla_V":[{}],"tesla_alerts":[{"Config":{"verbose":true}}],"tesla_errors":[{"Config":{"verbose":true}}]}},"activity":true,"context":"fleet-telemetry","level":"info","msg":"InitServer","time":"2024-10-02T16:03:16Z"}
app-1         | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"MUX","mux":{},"time":"2024-10-02T16:03:16Z"}
app-1         | Failed to obtain reader, failed to marshal fields to JSON, json: unsupported type: chan *telemetry.Record
app-1         | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"C ReliableAcksources","reliableAcksources":null,"time":"2024-10-02T16:03:16Z"}
app-1         | {"activity":true,"airbrake":{},"context":"fleet-telemetry","level":"info","msg":"AirbrakeHandler","time":"2024-10-02T16:03:16Z"}
app-1         | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"Port","server_port":8443,"time":"2024-10-02T16:03:16Z"}
app-1         | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"Host","server_host":"localhost","time":"2024-10-02T16:03:16Z"}
app-1         | Failed to obtain reader, failed to marshal fields to JSON, json: unsupported type: http.HandlerFunc

[image]

I can't fetch the live streaming data.
Please let me know what is wrong.

agbpatro, 2024-10-02:
Hi there
the errors you see are due to logging of objects like `server`, `mux` etc. Lorgus doesn't work well with them. See https://github.com/sirupsen/logrus/issues/642#issue-261749813.

Few more observations:-
```ssh
{"SocketServer":{"DispatchRules":{"tesla_V":[{}],"tesla_alerts":[{"Config":{"verbose":true}}],"tesla_errors":[{"Config":{"verbose":true}}]}},"activity":true,"context":"fleet-telemetry","level":"info","msg":"InitServer","time":"2024-10-02T16:03:16Z"}
```

tesla_V":[{}] shouldn't be empty. Are you sure you are using a correct config for starting up the service? Sample config file
https://github.com/teslamotors/fleet-telemetry/blob/main/examples/server_config.json

dorrod-blip, 2024-10-02:
@agbpatro, Thanks for your reply.
This is my config.json

`{
    "host": "0.0.0.0",
    "port": 8443,
    "log_level": "info",
    "json_log_enable": true,
    "reliable_ack": false,
    "namespace": "tesla_telemetry",
    "monitoring": {      
      "prometheus_metrics_port": 9090,
      "profiler_port": 4269,
      "profiling_path": "/tmp/trace.out"    
    },
    "logger": {
      "verbose": true
    },
    "kafka": {
        "bootstrap.servers": "kafka:9092",
        "queue.buffering.max.messages": 1000000,
        "queue.buffering.max.ms": 50,
        "compression.type": "gzip"
    },
    "rate_limit": {
      "enabled": true,
      "message_interval_time": 30,
      "message_limit": 1000
    },
    "records": {
      "tesla_alerts": [
          "logger"
      ],
      "tesla_errors": [
          "logger"
      ],
      "tesla_V": [
          "kafka"
      ]
    },
    "tls": {
      "server_cert": "/home/ec2-user/cert/tls-cert.pem",
      "server_key": "/home/ec2-user/cert/tls-key.pem"
    }
}`

Please let me know what is wrong. I know the errors are occured in mux and server.
First of all, I need to confirm if this is working or not `mux.HandleFunc("/", socketServer.ServeBinaryWs(c))`.
According to logs, I can't see the c values in logs because of failed.

dorrod-blip, 2024-10-02:
And when installing the Vehicle Command SDK, I used the registered TLS certificate and key in fleet telemetry.

closed 2024-10-03 by dorrod-blip