teslamotors/fleet-telemetry #221: zmq issue

ciumagcatalin, 2024-10-04:
Hello,

I am trying to setup the ZQM integration for dispatcher and I am getting this issue. I think the config piece is not right and there is not much information on how to set it:
Can somebody that did this before check this setup?

 "port": 443,
  "log_level": "debug",
  "json_log_enable": true,
  "namespace": "telemetry",
  "reliable_ack": false,
  "rate_limit": {
    "enabled": false,
    "message_limit": 100
  },
  "pubsub": {
       "gcp_project_id": "tesla-telemetry-437316"
  },
  **"zmq": {
       "addr": "tcp://xxxxxx.compute-1.amazonaws.com:5555"
  },**
  "records": {
    "alerts": [
        "logger",
        "pubsub"
    ],
    "errors": [
        "logger",
        "pubsub"
    ],
    "V": [
         "logger",
         "pubsub",
         **"zmq"**
    ]
  },

pp-1  | 2024/10/04 15:30:20 maxprocs: Leaving GOMAXPROCS=1: CPU quota undefined
app-1  | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"config_skipping_empty_metrics_provider","time":"2024-10-04T15:30:20Z"}
app-1  | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"starting_server","time":"2024-10-04T15:30:20Z"}
app-1  | {"activity":true,"context":"fleet-telemetry","level":"info","msg":"pubsub_registerd","namespace":"telemetry","project":"tesla-telemetry-437316","time":"2024-10-04T15:30:20Z"}
app-1  | panic: no such device
app-1  |
app-1  | goroutine 1 [running]:
app-1  | main.main()
app-1  |        /go/src/fleet-telemetry/cmd/main.go:46 +0x125

Catalin

Bre77, 2024-10-04:
Sorry I didn't get back to you on Discord, I couldn't remember how it had it configured but I've gone back through my commit history now.

You configure the Fleet Telemetry server as a ZMQ producer, which ZMQ clients connect to, so you shouldn't put an external domain, but rather an IP and port to bind to.

`"zmq":"0.0.0.0:3000"`

ciumagcatalin, 2024-10-04:
> Sorry I didn't get back to you on Discord, I couldn't remember how it had it configured but I've gone back through my commit history now.
>
> You configure the Fleet Telemetry server as a ZMQ producer, [...]

> Sorry I didn't get back to you on Discord, I couldn't remember how it had it configured but I've gone back through my commit history now.
>
> You configure the Fleet Telemetry server as a ZMQ producer, [...]

No worries. Thank you for the input here! More people might benefit from the guidance.
On the topic: I added as in the config: "zmq": {
       "addr": "tcp://0.0.0.0:4444"
  },
Error disappeared but the subscriber that is outside of the server does not receive the messages. Logs look clean on the telemetry, but I did not install ZMQ on it (Amazon Linux 2023 distribution). Two questions here:
1) Does telemetry comes with zmq library components for itself to function or it needs to be installed? I assume it would log the errors if not.
2) If telemetry becomes ZQM producer then I should be able to be subscriber on same instance as well. Do I get this right?

Bre77, 2024-10-05:
The way I used it was I had my NodeJS application connect to my fleet telemetry server using ZMQ. You dont need to install any other software to use ZMQ as its connections are directly client to server.

You need to configure your subscriber to connect to the IP of your fleet telemetry container/server on 4444 (your exposing that port right?)

patrickdemers6, 2025-01-22:
Closing this as it's gone stale. Please re-open if still facing this issue.

closed 2025-01-22 by patrickdemers6