teslamotors/fleet-telemetry #223: What does this error mean?

lizhanghua945, 2024-10-16:
2024-10-16 14:35:09,342 I  [TeslaHttpUtil       ] -353- param: {"config":{"hostname":"telemetry.test.etc.hrhaoche.com","port":443,"exp":1700000000,"fields":{"field":"DricveRail","value":{"interval_seconds":60}},"ca":"-----BEGIN CERTIFICATE-----\nMIICkzCCAXsCCQDxqHKtAyFjMTANBgkqhkiG9w0BAQsFADAqMSgwJgYDVQQDEx90\nZWxlbWV0cnkudGVzdC5ldGMuaHJoYW9jaGUuY29tMB4XDTI0MTAxNjAzNTEzNFoX\nDTI1MTAxNjAzNTEzNFowcTELMAkGA1UEBhMCVVMxEzARBgNVBAgMCkNhbGlmb3Ju\naWExFjAUBgNVBAcMDVNhbiBGcmFuY2lzY28xGDAWBgNVBAoMD0ZsZWV0IFRlbGVt\nZXRyeTEbMBkGA1UEAwwSRmxlZXQgVGVsZW1ldHJ5IENBMIGfMA0GCSqGSIb3DQEB\nAQUAA4GNADCBiQKBgQCupQrPklalQcPPug6E54aEP/GtOc6vWsypPJPAagS7nSVJ\ned/kDWXnh8EuzbLrRF9Cr+gggGZgrVkwUt2qbT6UGFXcx4Zo47Dl18HTGqSS4pts\ncpCAyAdN4qegelAtY9dyy/iH9Asw1UjVjd3g3zUs/2f7KGRa2YX9VPbxiWdCyQID\nAQABMA0GCSqGSIb3DQEBCwUAA4IBAQBEY2HRyo2q9r5QQqu4ilW0dg9AkLphsAYp\ncmRIBXDN2Bv4oAL+iNsZLizjexcdYxt0oA1Vl0XJ3pyIupyk3s0rAdXXHZkcAarV\nPoVyWYSJ9n94/1Yiz8kIYdrS5q8CdYb+bF7mbb+60ny0Fjkrn/DZKI7ebUTIT35B\nWOnTn3LYMUIr2hA2pM2ybouy5/LuESCuoprSb3T/oABIqOrJSczRr4hTUlmq5FJm\nplrWzDrlHNrTt7QJ8moJheK8hYMCpO93Zfkwzi4W7RxnhW3oZ1OqIi8kxx0FV0tq\nKiDcicTIWOO0G1mp4ryK+tBiFoBzK99j4I6Mwf+CpM499exvDCgv\n-----END CERTIFICATE-----\n"},"vins":["LRW3E7ET3RC217335"]}
2024-10-16 14:35:09,343 I  [TeslaHttpUtil       ] -357- httpPost: POST https://172.17.239.105/api/1/vehicles/fleet_telemetry_config HTTP/1.1
2024-10-16 14:35:09,675 I  [TeslaHttpUtil       ] -361- Response body: {"response":null,"error":"Refer to https://developer.tesla.cn/docs/fleet-api/fleet-telemetry for instructions setting up fleet-telemetry.","error_description":"","txid":"3bfce9ff41c21cb5a5a96838222951a6"}
2024-10-16 14:35:09,676 I  [TeslaControl        ] -397- ret:{"response":null,"error":"Refer to https://developer.tesla.cn/docs/fleet-api/fleet-telemetry for instructions setting up fleet-telemetry.","error_description":"","txid":"3bfce9ff41c21cb5a5a96838222951a6"}

rileymd88, 2024-10-16:
You pasted your private key here - you should remove it. Also you
misspelled the field name in your config.

lizhanghua945, 2024-10-17:
thanks,now is ok {"response":{"updated_vehicles":1}}

closed 2024-10-17 by lizhanghua945