teslamotors/fleet-telemetry #226: How to get values from new data fields such as DetailedChargeState?

andrewdevelopz, 2024-10-25:
using:
**fleet-telemetry - v0.3.2**

Hello,

Can someone please help me on how I can get the values for the [DetailedChargeState](https://github.com/teslamotors/fleet-telemetry/blob/main/protos/vehicle_data.proto#L191) field?

So, I send a post request to [fleet-telemetry-config-create](https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-endpoints#fleet-telemetry-config-create), with `DetailedChargeState` included in `fields` property.

Using [fleet-telemetry-config-get](https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-endpoints#fleet-telemetry-config-get), I see that the vehicle is synced and `DetailedChargeState` is included in `fields` property.

The fleet-telemetry service is successfully receiving payloads and inserting them into kafka. I inspect the gathered data from kafka by looping through the data and using `$value->hasDetailedChargeStateValue()` method provided by [google/protobuf](https://packagist.org/packages/google/protobuf). It returns false on every loop.

Also `$value->getDetailedChargeStateValue()` doesn't have the proper value

```php
$payload->mergeFromString($message->payload);

$vehicleData = $payload->getData();

/** @var Datum $item */
foreach ($vehicleData as $item) {
    $key = $item->getKey();
    $value = $item->getValue();

    $value->hasDetailedChargeStateValue(); // `false` on every loop

    $value->getDetailedChargeStateValue(); // `0` on every loop
}
```

I have also tried checking for these new fields from the methods below but they also return false.

```
$value->hasShiftStateValue()
$value->hasDoorValue()
$value->hasSentryModeStateValue()
```

Thank you.

itsMeDavidV, 2024-10-31:
Same. Can confirm `DetailedChargeState` is included in my vehicle's synced config, however no kafka messages contain `DetailedChargeStateValue` or a string value representing `DetailedChargeStateValue`.

Is there something we are missing?

This also seems to be the only field that we can except to return typed values, maybe an issue with returning typed values?

> // fields below here are always returned typed
>  DetailedChargeState = 179;

dominicboston-kaluza, 2024-11-06:
Just wondering if anyone has made progress with this or worked out how to use this new field? Still seems to be returning 179?

rmgodbole, 2024-11-09:
2024-11-09 03:36:09 tesla-telemetry-1        | {"activity":true,"context":"fleet-telemetry","data":[{"Body":"","CreatedAt":1731139143,"Name":"unsupported_field","Tags":{"field_name":"DetailedChargeState","name":"35c4f7b5ae30-4d46-8ad3-688501fe2091"}}],"level":"info","metadata":{"receivedat":"1731152169000","timestamp":"0","txid":"","txtype":"errors","version":"0","vin":"<<<VIN>>>"},"msg":"record_payload","time":"2024-11-09T11:36:09Z","vin":"<<<VIN>>>"}

**fleet_telemetry** v0.3.5 
**firmware** v12.5.4.1

patrickdemers6, 2024-11-13:
`DetailedChargeState` is available beginning with firmware version `2024.38`. Once your vehicle is updated to this version, you will begin receiving the field.

closed 2024-11-13 by patrickdemers6

andrewdevelopz, 2024-11-13:
@patrickdemers6 Ah I see, thank you I appreciate the response.