teslamotors/fleet-telemetry #230: Connectivity topic cannot parse invalid wire-format data

Bre77, 2024-10-30:
Just enabled  the new connectivity topic, but am getting these errors and no connectivity events.

```
{"context":"fleet-telemetry","error":"proto: cannot parse invalid wire-format data","level":"error","metadata":{"receivedat":"1730252498000","timestamp":"0","txid":"4bba9ebc-f4a3-48a8-8475-cb603a784fa4","txtype":"connectivity","version":"0","vin":"redacted"},"msg":"record_logging_error","time":"2024-10-30T01:41:38Z","vin":"redacted"}
```

Relevant config
```json
{
    "log_level": "info",
    "json_log_enable": true,
    "transmit_decoded_records": true,
    "reliable_ack": false,
    "namespace": "na",
    "records": {
        "alerts": [
            "pubsub"
        ],
        "connectivity": [
            "pubsub"
        ],
        "errors": [
            "pubsub"
        ],
        "V": [
            "pubsub"
        ]
    }
}
```

Bre77, 2024-10-30:
I am starting to suspect that `connectivity` doesnt support `logger`.

agbpatro, 2024-10-30:
thanks. looks like setting `transmit_decoded_records` to `true` is not compatible with logger. I have submitted a pull request to fix the issue

closed 2024-10-30 by agbpatro