teslamotors/fleet-telemetry #241: Data update

krishramani01, 2024-11-26:
How does tessie updating data in real-time without using Telemetry ?

Bre77, 2024-11-26:
Tessie **does** use Fleet Telemetry if you let it (https://help.tessie.com/article/128-direct-telemetry). Otherwise they are just polling.

krishramani01, 2024-11-27:
I want to do same like tessie how many rate limit required for that? Does tesla increase my rate limit same as tessie ?

Bre77, 2024-11-27:
Tesla hasnt made the higher subscription levels public yet. In theory Tessie is paying Tesla for a higher rate limits, which is why they had to increase their prices so much recently.

krishramani01, 2024-11-27:
I paid same like tessie for higher rate limit so does tesla allow this ?

netdata-be, 2024-11-27:
It seems the rate limite of 10 fields/min has been lifted, I'm able to create something useful now....

krishramani01, 2024-11-27:
I want to develope app like tessie with real-time data update so could you tell how can I do it ?

netdata-be, 2024-11-27:
> I want to develope app like tessie with real-time data update so could you tell how can I do it ?

Do as the docs say you to do so ;-)
- Install the vehicle command proxy
- install the fleet-telemetry
- send data to one of the supported dispatchers
- Create your own software which consumes these data

krishramani01, 2024-11-27:
I already did it! Tesla has not added the climate field or valet mode status and many field missing so I cannot update it in real time.

netdata-be, 2024-11-27:
> I already did it! Tesla has not added the climate field or valet mode status and many field missing so I cannot update it in real time.

For these fields you can query the [vehicle_data](https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-endpoints#vehicle-data) endpoint, just make sure you implement it correctly so that you do not wake the vehicle up if not needed...

krishramani01, 2024-11-27:
True but api pooling that's not solution

netdata-be, 2024-11-27:
Why not, teslamate uses a combination of polling and pushes, agree the telemetry data does not contain ALL fields, but that is your use case to get telemetry data on the climate settings?

krishramani01, 2024-11-27:
> Why? You can query the API currently every 8 min without a problem. What is the use case of have more updates on the climate data?

When the user starts the climate control from their car, the app should provide real-time updates, showing the climate's start state immediately instead of after 8 minutes.

patrickdemers6, 2025-01-22:
We have added additional fields in recent firmware versions and are working on adding more. Going to close this as other issues detail specific asks.

closed 2025-01-22 by patrickdemers6