teslamotors/fleet-telemetry #244: BUG: Connectivity Events not reliable?

netdata-be, 2024-11-29:
Hi, 

When using the command-api I see my vehicle is going to sleep and is offline.

```
/api/1/vehicles/<VIN>  | jq .response.state
"offline"
```

I have enabled the connectivity to log to stdout, there I find these events:

```
{"ConnectionID":"f6433181-31d7-4831-aff3-78979bb019f2","CreatedAt":1732880726,"Status":"CONNECTED","Vin":"xxx"}
{"ConnectionID":"f6433181-31d7-4831-aff3-78979bb019f2","CreatedAt":1732880726,"Status":"DISCONNECTED","Vin":"xxx"}
{"ConnectionID":"0042e097-2122-414b-8dcf-417b5d41b200","CreatedAt":1732880726,"Status":"CONNECTED","Vin":"xxx"}
{"ConnectionID":"63b9e371-7d2b-4277-9b8e-96171a847b42","CreatedAt":1732881637,"Status":"CONNECTED","Vin":"xxx"}
```

A few questions here:
1. Why does the vehicle connect + reconnect at the same time?
2) I see 2 CONNECTED events, but none DISCONNECTED event
3) Why does the vehicle have 2 CONNECTED Events ?

Bre77, 2024-11-29:
I've also found these events to be unreliable. I was really looking forward to having a strong realtime awake vs asleep/offline signal, but am seeing similar behaviour from my Model 3.

[image]

stx, 2024-12-06:
Yeah, looking at the times when data is sent by Fleet Telemetry (along with the car timestamps of those data points), and looking at when Fleet Telemetry reports Connected/Disconnected, there is basically no correlation.

While "Disconnected", data is still sent and the car is still awake and active. Happens across all models. There's no way this can be used for sleep/wake detection as is. Might need another solution.