teslamotors/fleet-telemetry #251: ScheduledChargingStartTime has no time zone

jbanyer, 2024-12-03:
The field `ScheduledChargingStartTime`, when converted to JSON, has no time zone. It appears to be in the local time zone of the vehicle?

```
    {
      "key":"ScheduledChargingStartTime",
      "value":{
        "stringValue":"Wed Dec 4 00:15:00 2024"
      }
    },
```

In the polling API `scheduled_charging_start_time` is a Unix time.

Can we get `ScheduledChargingStartTime` in a format suitable for parsing including a timezone? Eg Unix time or ISO timestamp?

Thanks!

jbanyer, 2025-01-12:
In addition: when `prefer_typed` is true in the telemetry config, the data looks like this:
```
         "key":"ScheduledChargingStartTime",
         "value":{
            "longValue":"1736756100"
         }
```

The time zone appears complete wrong here. Unix time 1736756100 is equal to:
- GMT: Mon Jan 13 2025 08:15:00 GMT+0000
- local time: Mon Jan 13 2025 19:15:00 GMT+1100

The schedule start time is actually 00:15, which doesn't match either of the above interpretations of the unix timestamp. It appears as though a time zone of GMT-8 has been used. LA time zone - no coincidence??

When `prefer_typed` is false the data looks like this:
```
   "key":"ScheduledChargingStartTime",
   "value":{
      "stringValue":"Mon Jan 13 00:15:00 2025"
   }
```

jbanyer, 2025-01-16:
@ThomasAlxDmy any chance of a comment here? This is preventing us from using `prefer_typed`, and thus the new `minimum_delta` config option. Thanks!

jbanyer, 2025-01-17:
@patrickdemers6 any chance of a comment here? This is preventing us from using `prefer_typed`, and thus the new `minimum_delta` config option. Thanks!

patrickdemers6, 2025-01-17:
Sorry for this @jbanyer, this is a bad bug.

The vehicle represents ScheduledChargingStartTime in its local timezone, but then the values are being shifted and reported in pacific, as you said.

I can work on a firmware fix for this. It will technically be breaking as it changes behavior, but I'd argue it's completely broken as is.

jbanyer, 2025-01-18:
@patrickdemers6 thanks for your reply. I can work around the bug until it is fixed, by interpreting the epoch time as `America/Los_Angeles` time zone.

Can you please update this ticket when you know the firmware version that includes the fix?

Cheers.

patrickdemers6, 2025-01-22:
Yes, will do. Also planning to introduce a better way to communicate the version of Fleet Telemetry running on the vehicle. As I'm sure you've noticed, our branching/releases are complicated.

jbanyer, 2025-02-12:
@patrickdemers6 any update here? Is the prefer_typed value (eg `"longValue":"1736756100"`) still in LA timezone for all vehicles?

patrickdemers6, 2025-02-12:
No firmware changes have been made related to this. Mind reviewing #359 which applies a server side fix?

jbanyer, 2025-02-12:
@patrickdemers6 I reviewed #359 and made one comment about daylight savings.

To confirm, with #359:
- timestamps should be interpreted as UTC, and the result will actually be in the local time of the car (for `ScheduledChargingStartTime` at least)
- the fleet-telemetry process will handle different transformations for different firmware versions, if required in future