teslamotors/fleet-telemetry #252: Data fields and support

Urkman, 2024-12-03:
Hello devs @Tesla @patrickdemers6 @sethterashima and others,

now that Tesla wants us to pay for the data, the support needs to be enhanced dramatically.
Please answer ALL open questions here and also answer emails...

We need to pay the price, so we need all the information and answers...

And some questions about data
You want us to switch to Telemetry? Fine then give us all the data we need.
Here are some fields that I'm currently missing.

Climate on / off
Defrost on / off
Steeringwheelheater on / off
Chargeport opne / closed
Preset Climate temp
ClimateKeepMode
and I think, there are many other missing fields. Perhaps someone wants to add field here ;) 

Thanks,
Stefan

Bre77, 2024-12-03:
Regarding the missing fields, it looks like they are adding a bunch of missing ones in the Christmas update.

https://github.com/teslamotors/fleet-telemetry/pull/248

Urkman, 2024-12-03:
Fine, that covers at least some of the fields...

Urkman, 2024-12-03:
As I see, that leaves this fields open for my list:
Climate on / off
Steeringwheelheater on / off
Preset Climate temp

But perhaps I miss something?

Bre77, 2024-12-03:
> As I see, that leaves this fields open for my list:
> Climate on / off
> Steeringwheelheater on / off
> Preset Climate temp
> 
> But perhaps I miss something?

I'd expect those to be:
ClimateACEnabled
HvacSteeringWheelLevel
HvacLeftTemperatureRequest (and right)

I'm sure there are still some missing, but I haven't done the work to figure that out yet.

junghoon-vans, 2024-12-03:
Dear @teslamotors Team,

I am writing to inquire about the support status of certain fields in the Fleet Telemetry API protocol.

We have been using several fields in the Vehicle Data API, and I would like to verify if these fields are still supported in the Fleet Telemetry API, or if they have been renamed:

1. User/Security Related
   - `is_user_present` (detects user presence in vehicle)
   - `valet_pin_needed` (indicates if valet PIN is required)

2. Interior Convenience Related
   - `has_seat_cooling` (ventilated seats support)
   - `sun_roof_installed` (sunroof installation status)

As these fields are still essential for our implementation, could you please guide us on how to access this information through the Fleet Telemetry API?

Thank you for your assistance.

Best regards,

Urkman, 2024-12-03:
And one more point here...
**cruise_state** was removed again without any comment.
That's the kind of stuff that should **never** happen :(

patrickdemers6, 2025-01-22:
> the support needs to be enhanced dramatically.
We're doing our best to handle the support load and are working on improving our internal processes.

> Also reliable **Gear** would be cool. In the past we had **Gear**=P with **VehicleSpeed**>0. Drifting? :D

Tracking the occasional dropped signal in #319 

---

@junghoon-vans noted these as additional fields to add. Re: other field requests, they should be covered in recent firmware releases.

closed 2025-01-22 by patrickdemers6