teslamotors/fleet-telemetry #253: Field for software version?

jbanyer, 2024-12-03:
Is there a telemetry field for the vehicle's software version?

The `Version` field just contains a space for me. Here is the JSON format:

```
{"key":"Version", "value":{"stringValue":" "}}
```

netdata-be, 2024-12-03:
There will be, the holiday update will bring us this data.
https://github.com/teslamotors/fleet-telemetry/pull/248/files#diff-5a5dfc53aca1261e04717e3d5fbeee2135ecee91a2e9ee9178e2253880c305f4R236

So I think you can close this one ;-)

corsair, 2024-12-04:
> There will be, the holiday update will bring us this data. https://github.com/teslamotors/fleet-telemetry/pull/248/files#diff-5a5dfc53aca1261e04717e3d5fbeee2135ecee91a2e9ee9178e2253880c305f4R236
> 
> So I think you can close this one ;-)

The Version they are referring to is separate from the new fields added in #248. The new software update fields added refer to a software updates pending installation/download.

Thus, the issue causing this field to be empty/a space is still a concern.

patrickdemers6, 2024-12-05:
I'll be publishing descriptions for all fields soon-ish.

`Version` is intended to be the version the vehicle is running. Up until 2024.44.25, it returns the version of an available software update, which is why it is typically empty. This was a bug that has been present since fleet telemetry was introduced.

Starting with .44.25, `Version` properly returns the version the vehicle is running. The newly introduced field `SoftwareUpdateVersion` will return the software version available for update.

davidsl4, 2024-12-05:
> I'll be publishing descriptions for all fields soon-ish.

That would really save us a huge amount of time! Thank you @patrickdemers6!

jbanyer, 2025-01-07:
Closed as this is now appearing eg:
```
      {
         "key":"Version",
         "value":{
            "stringValue":"2024.44.25.4 ad52e3119f65"
         }
      },
```

closed 2025-01-07 by jbanyer