teslamotors/fleet-telemetry #255: Difference betwee ACChargingEnergyIn and DCChargingEnergyIn

dirkbonhomme, 2024-12-04:
I am currently charging my car with an AC charger. These are the latest values:
```
ACChargingEnergyIn: 16.24488913095765
DCChargingEnergyIn: 15.659999649971724
```
Does anyone know the exact meaning of both values? My guess is that ACChargingEnergyIn is the energy that is entering the car's charger and DCChargingEnergyIn is the energy that is eventually entering the battery?

Even so, how can I detect whether the car is AC or DC charging?

davidsl4, 2024-12-05:
I’m no expert, but it seems like you might be onto something. If the `ACChargingEnergyIn` shows the energy from the charger and the `DCChargingEnergyIn` reflects the energy actually added to the battery, then I guess you could compare those values. Note it's possible that during DC charging, `ACChargingEnergyIn` could also drop to zero. There's nothing that trial and error won't answer, so I would advise exploring both options.

And I came across another field that could be useful: `FastChargerPresent`.

rawmean, 2024-12-30:
I think the first one is supposed to be the energy added during AC charging (eg, home charging) and the 2nd one is the energy added during supercharging (which is DC charging). 

the problem is that the `ACChargingEnergyIn` value does not match what the Tesla app reports under "charging stats"

Adminius, 2025-01-24:
We are using now `DCChargingEnergyIn `while AC Charging.
In my understanding is DC = added to battery after all possible losses (AC->DC, heater and so on)
`ACChargingEnergyIn` is AC side, so before AC->DC losses, heater and so on.