teslamotors/fleet-telemetry #261: How to Unregister an Application for a Specific Region and Domain?

EmilOlovsson, 2024-12-09:
I’m looking for a way to remove the registration of an application tied to a specific domain and region.

Example URL:
curl -H "Authorization: Bearer $TESLA_AUTH_TOKEN" \
-i https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts/public_key?domain=MY_DOMAIN_URL

The goal is to re-register __MY_DOMAIN_URL__ for a new application. Despite reaching out to fleetapisupport@tesla.com multiple times over the past few months, I have not received a response.

Has anyone managed to remove an application or un-register it from a domain?

Bre77, 2024-12-09:
Ive done this before by changing the partner domain of the old application to something else, which make my actual domain avaliable to register by the new application.

EmilOlovsson, 2024-12-09:
> Ive done this before by changing the partner domain of the old application to something else, which made my actual domain avaliable to register by the new application.
> 
> But this was back in Feburary, so hopefully it still works.

I tried this and I managed to register my new application on region https://fleet-api.prd.na.vn.cloud.tesla.com/.
I generated a new EC Private key and hosted the public key at my domain. The problem now is that my I can't register my domain on region https://fleet-api.prd.eu.vn.cloud.tesla.com/ with my new public key.

{"response":null,"error":"Validation failed: Csr public key does not match the com.tesla.3p.public-key.pem","error_description":"","txid":"xxxxx"}

Any ideas how to fix this?

Bre77, 2024-12-09:
Did you change the old applicaiton registration on both NA and EU? Rememeber you have to do everything twice, once on each region.

EmilOlovsson, 2024-12-09:
> Did you change the old applicaiton registration on both NA and EU? Rememeber you have to do everything twice, once on each region.

These are the steps I have done so far.

1. Application 1 - Registered to EU. I Want to register it to both EU & NA.
2. Application 2 - Registered to NA. I want to remove this application or un-register this.

What I have done.
1. Application 2 - Registered to a new domain in NA.
2. Application 1 - Tried to register to NA. Result: {"response":null,"error":"Validation failed: Public key hash has already been taken","error_description":"","txid":"607ac89a9416885d99e8ee8269845001"}
3. Generated a new EC public key.
4. Application 1 - Registered to NA. Success!
5. Application 1 - Register to EU. Error! {"response":null,"error":"Validation failed: Csr public key does not match the com.tesla.3p.public-key.pem","error_description":"","txid":"20c40ff04bb6c3fa6776590c0dca991f"}

Hopefully this clarifies the steps I have taken and what issues I'm encountering.

Bre77, 2024-12-09:
It does, you haven't changed app 2 domain in the EU.

EmilOlovsson, 2024-12-09:
> It does, you haven't changed app 2 domain in the EU.

I registered Application 2 to EU too and then tried to register Application 1 to NA (succcess) & EU (failure).

{"response":null,"error":"Validation failed: Csr public key does not match the com.tesla.3p.public-key.pem","error_description":"","txid":"fb376e1e6b8b78bd20b8e65626dc0fb7"}

EmilOlovsson, 2024-12-18:
@caglarsu No, I haven't solved the problem yet. Let me know if you find a way.

patrickdemers6, 2025-01-22:
We've removed backend validations so you should be able to change keys now.

closed 2025-01-22 by patrickdemers6

gbertsch, 2025-03-01:
@patrickdemers6 - Having similar issue to OP, had to change my public key a few times (was debugging), now perhaps I have exhausted some limit? As I am unable to re-register my new public key in 'na' region. As a workaround, I went to register and use 'eu' domain to distribute virtual keys.

Any further thoughts around this error?

{"response":null,"error":"Validation failed: Public key hash has already been taken","error_description":"","txid":"dc3f6d509c6f70f2f6b57b3dd79437a1"}%