teslamotors/fleet-telemetry #263: Vehicle Alias on List

KarlF-KFLOE, 2024-12-10:
It seems like vehicle alias was removed from the vehicle list endpoint.  Can we get it back please?

bassmaster187, 2024-12-10:
What is a "Vehicle Alias"?

KarlF-KFLOE, 2024-12-10:
Sorry, display_name.  It is coming back as null but used to be a string containing the friendly name of the vehicle.

Bre77, 2024-12-10:
I believe this is a known issue. Its still avaliable on the products endpoint, and is also avaliable in the `vehicle_data` endpoint under vehicle_state.vehicle_name

KarlF-KFLOE, 2024-12-11:
Thanks Brett.  We're trying to avoid the cost of calling vehicle_data whenever possible and the energy scopes aren't needed for our app so we can't fallback to products.  I appreciate the suggestions though.

I searched the issues here before posting, but did not find anything.  Is there a known issues log I should check?

bassmaster187, 2024-12-12:
It's VehicleName and works well:
{ "key": "VehicleName", "value": { "stringValue": "Tessi" }

rawmean, 2024-12-13:
I get this from telemetry (the car does have a name):
`{'key': 'VehicleName', 'value': {'invalid': True}`

ThomasAlxDmy, 2025-01-16:
Should be back now - can you confirm?

rawmean, 2025-01-16:
> Should be back now - can you confirm?

It's not back for cars that are on 2024.44.25.2 and older (which is most of the fleet). Do we have to wait for those cars to get newer firmware?

KarlF-KFLOE, 2025-01-16:
@ThomasAlxDmy any plan to add it back on the vehicle list endpoint?

patrickdemers6, 2025-01-22:
display_name is back, for real this time!

closed 2025-01-22 by patrickdemers6

KarlF-KFLOE, 2025-01-22:
@patrickdemers6 confirmed and thank you!!!