teslamotors/fleet-telemetry #264: Kinesis setup guide

dirkbonhomme, 2024-12-11:
I'm stuck connecting the telemetry service with a Kinesis stream.

I have installed fleet-telemetry with Helm and am receiving signals. However using both the local Kinesis (https://github.com/saidsef/aws-kinesis-local) and a stream hosted at AWS results in an error when the service receives a signal:

```	
2024/12/11 16:32:44 http: panic serving xxx.xx.xx.xx:36010: runtime error: invalid memory address or nil pointer dereference
goroutine 84 [running]:
net/http.(*conn).serve.func1()
	/usr/local/go/src/net/http/server.go:1854 +0xbf
panic({0x10692a0, 0x18d7950})
	/usr/local/go/src/runtime/panic.go:890 +0x263
github.com/teslamotors/fleet-telemetry/telemetry.(*BinarySerializer).Dispatch(...)
	/go/src/fleet-telemetry/telemetry/serializer.go:93
github.com/teslamotors/fleet-telemetry/telemetry.(*Record).Dispatch(0xc000000900)
```

Config:

```{
   ...
  "kinesis": {
    "override_host": "http://kinesis:4567",
    "max_retries": 3,
    "streams": {
      "V": "tesla-telemetry-development"
    }
  },
  "records": {
    "alerts": [
      "logger"
    ],
    "errors": [
      "logger"
    ],
    "V": [
      "logger",
      "kinesis"
    ]
  },
  ...
}
```

Environment for fleet-telemetry deployment:

[image]

Not sure what I'm missing?

dirkbonhomme, 2024-12-11:
Nevermind, the Helm chart is still using Docker image v0.0.3. This is sloppy Tesla 😬

dirkbonhomme, 2025-01-16:
@ThomasAlxDmy I've reopened this ticket because the Helm chart is still referring to an old version of Fleet Telemetry. We were able to figure it out eventually 😅 but anyone new might run into the same issues when following the README.

agbpatro, 2025-01-16:
Hi @dirkbonhomme 
We published a new version of helm-charts couple of weeks ago https://github.com/teslamotors/helm-charts/releases/tag/fleet-telemetry-0.1.1
Can you try using that?

closed 2025-01-16 by agbpatro